micropython/micropython #4836: Want to try out Teensy 3.5, need advice

mindforger, 2019-06-07:
since i have done my experiments with NUCLEO Boards and being rather unsatisfied with the lack of direct USB access (besides soldering on a cut wire)

i wanted to move ahead to look for compackt DIP style boards that can run uPy

i opted out for the pyboards as they do not have a DIP style pin layout (i could leave the lower end  pins unpopulated but the board still is a bit wide to stuff it on a breadboard)

the new pyboard-d has way too much stuff on it to use it efficiently for me

but the teensy 3.5 seems okay with me 

i have found the teensy in the wiki but i can not find any support source wise except for a merge request 
https://github.com/micropython/micropython/pull/2513

which seems to not be finished yet

so i need advice (lack of git command line understanding here)  on how to grab the branch from @dhylands to compile it myself and give it a test

dhylands, 2019-06-07:
Another board to look at is the 1-Bitsy: https://1bitsy.org/ 

The directory structure changed quite a bit, so I'd recommend checking out into a new directory using a command like:
```
git clone --single-branch --branch teensy-3.6 https://github.com/dhylands/micropython upy-teensy
cd upy-teensy/teensy
make BOARD=TEENSY_3.5
```

robert-hh, 2019-06-07:
IT's a pity that Teensy 3.5 and 3.6 are not supported in the main line, because these are good boards. Besides that, you can use a PyBoard on a breadboard, if you solder female with long pins headers, such that you can connect wires at the top and stick the Pyboard to the breadboard, still being able to place jumpers below the PyBoard.

mindforger, 2019-06-07:
i hope i can do some testing and improvement on my own :) Thank you for the help, i have an issue to understand what part of the pullrequest-page is important to enter in the git line ... 

but as i see now dhylands:teensy-3.6 means the repository is located under .com/dhylands instead of .com/micropython and teensy-3.6 is the actual branch inside THAT fork ... that also explains why i could never recreate my first checkout for the L432KC because i always tried to use a non exsitting branch from the main and not the fork :D (after i used your fresh checkout from your fork directly it worked)

closed 2019-06-07 by mindforger

dhylands, 2019-06-07:
It is also possible to checkout a PR oin the main repository. I use this script: https://github.com/dhylands/dotfiles/blob/master/bin/git-pr originally from: https://gist.github.com/jhnns/d654d9d6da6d3b749986

You put git-pr someplace in your path and then you can checkout a PR.

dhylands, 2019-06-07:
I'd like to get my teensy branch resynced with the micropython mainline - its just a matter of time. Damien wanted it to be a Kinetis port that happened to have teensy boards, rather than a teensy port, which is where things were left off.

dpgeorge, 2019-06-19:
> a Kinetis port that happened to have teensy boards, rather than a teensy port,

Yes, I'm still in favour of doing this.

mindforger, 2019-09-03:
i am getting my teensy 4.0 tomorrow (i skipped the 1Bitsy because it was always out of stock)

what is meant with that "Kinetis" Port thingy?! Is there some sort of code base where i should pick up the red thread to work myself into it? 
Or should i use the teensy port of @dhylands for temporary workaround?