micropython/micropython #4837: Pyboard D: I2C pullups are not enabled

peterhinch, 2019-06-08:
Checked on an SF2W and an SF6W, using a WBUS-DIP28 adaptor.

If I2C(2) is instantiated, pins Y9 and Y10 are pulled up. However the pullup is weak (presumably that internal to the STM chip): a 10KΩ resistor to Gnd pulls the voltage down to around 0.65V. This suggests that the drive to the onboard 3KΩ resistors is not enabled when the I2C object is created.

chuckbook, 2019-06-08:
Pull up resistors R16 & R17 are 5.6k connected to a  PH5 & PF1. It allows to control pull-ups for I2C1 (X9 & X10). Y9 & Y10 don't have any resistors.

robert-hh, 2019-06-08:
Too sad that X9 and X10 are not broken out at the  WBUS-DIP28 adapter.

peterhinch, 2019-06-09:
@chuckbook You are right. Odd that pullups aren't provided for I2C(2) too.

dpgeorge, 2019-06-12:
> Too sad that X9 and X10 are not broken out at the WBUS-DIP28 adapter.

WBUS-DIP28 is intended to be small (same length as PYBD), and since X1-X12 are relatively easy to access from the top using 0.1" headers it was Y1-Y12 which were taken out to the DIP28 pins.  So that gives X1-X12 and Y1-Y12.  That's the intention.

> Odd that pullups aren't provided for I2C(2) too.

The external ones are only on I2C('X') (due to routing constraints).  The weak internal STM pull-ups are usually enough, but otherwise you'll need to provide them yourself for I2C('Y').

closed 2019-06-12 by dpgeorge

robert-hh, 2019-06-12:
> WBUS-DIP28 is intended to be small (same length as PYBD), and since X1-X12 are relatively easy to access from the top using 0.1" headers it was Y1-Y12 which were taken out to the DIP28 pins. So that gives X1-X12 and Y1-Y12. 

This is not the same. Using the solder pads on the sides is a on the board requires a hard wired connection or at least soldered pins, while using the connectors at the bottom is easier. One can do both, but hardly at the same time.

dpgeorge, 2019-06-14:
> This is not the same.

Yes I understand, it does require soldering something to the PYBD itself, but these could be female header sockets so they can be reused by plugging in different male-ended wires (like on PYBv1.x).  At least the option is there.

If you have suggestions for an improved break-out board (eg with ~40 pins) I'd be keen to hear.

peterhinch, 2019-06-15:
I'm engaged in a discussion with @jimmo on this topic at the moment. The aim is to design something smaller than the DIP68 but providing a larger pin set than the DIP28, hopefully at lower cost by eliminating some functionality of the DIP28, namely power control and USB.

The discussion is converging on two options. Both (like the DIP68) have no components on the rear of the board. Both have a single tile header, and bring out all the Pyboard D pins on a 0.1" pitch. Pins are grouped in "natural" order with X pins together and in order, likewise Y pins. This facilitates creating a drawing for users showing the devices (SPI, I2C, CAN, etc.) as the pins for each are adjacent.

The 40 pin version is laid out such that stacking socket headers may be used on the Pyboard pins. This leaves 8 pins whose location precludes stacking headers (because of protrusions on the Pyboard). Their usage is TBD.

The 32 pin version is the same width as the DIP28 (I hope: this is under discussion). In the narrow option socket headers are not possible because of overhang of the Pyboard. In the wide option 8 pins would be overhung.

robert-hh, 2019-06-15:
@pythoncoder: That sounds good. Meanwhile, since I cannot solder the small 40 pin connectors with my 65+ years, I sketched a small adapter board, which hooks on female 0.05" pitch sockets soldered to the top of a PYBD. It carries the signals to two 0.1" pitch socket strips with 0.3" distance. Which means that the PYBD is mounted top-side-down, allowing to see the labels.

peterhinch, 2019-06-16:
A suggestion is that the board should be offered (as an option) with headers so it can be plugged into a breadboard. This opens up the Pyboard D to users lacking soldering skills. Stacking headers improve connectivity if the board overlays some breadboard holes.