micropython/micropython #4839: esp32 PPP compile errors

c0d3z3r0, 2019-06-08:
I am getting these errors when trying to compile micropython@master with IDF@5c88c5996

~~~
xtensa-esp32-elf-gcc -std=gnu99 -Os -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE='"mbedtls/esp_config.h"' -DHAVE_CONFIG_H -I../../lib/berkeley-db-1.xx/PORT/include -I. -I../.. -I../../lib/mp-readline -I../../lib/netutils -I../../lib/timeutils -Ibuild -I/home/c0d3/src/esp-idf.git/components/bootloader_support/include -I/home/c0d3/src/esp-idf.git/components/bootloader_support/include_bootloader -I/home/c0d3/src/esp-idf.git/components/driver/include -I/home/c0d3/src/esp-idf.git/components/driver/include/driver -I/home/c0d3/src/esp-idf.git/components/nghttp/port/include -I/home/c0d3/src/esp-idf.git/components/nghttp/nghttp2/lib/includes -I/home/c0d3/src/esp-idf.git/components/esp32/include -I/home/c0d3/src/esp-idf.git/components/soc/include -I/home/c0d3/src/esp-idf.git/components/soc/esp32/include -I/home/c0d3/src/esp-idf.git/components/ethernet/include -I/home/c0d3/src/esp-idf.git/components/expat/expat/expat/lib -I/home/c0d3/src/esp-idf.git/components/expat/port/include -I/home/c0d3/src/esp-idf.git/components/heap/include -I/home/c0d3/src/esp-idf.git/components/json/include -I/home/c0d3/src/esp-idf.git/components/json/port/include -I/home/c0d3/src/esp-idf.git/components/log/include -I/home/c0d3/src/esp-idf.git/components/newlib/platform_include -I/home/c0d3/src/esp-idf.git/components/newlib/include -I/home/c0d3/src/esp-idf.git/components/nvs_flash/include -I/home/c0d3/src/esp-idf.git/components/freertos/include -I/home/c0d3/src/esp-idf.git/components/esp_ringbuf/include -I/home/c0d3/src/esp-idf.git/components/esp_event/include -I/home/c0d3/src/esp-idf.git/components/tcpip_adapter/include -I/home/c0d3/src/esp-idf.git/components/lwip/lwip/src/include -I/home/c0d3/src/esp-idf.git/components/lwip/port/esp32/include -I/home/c0d3/src/esp-idf.git/components/lwip/include/apps -I/home/c0d3/src/esp-idf.git/components/mbedtls/mbedtls/include -I/home/c0d3/src/esp-idf.git/components/mbedtls/port/include -I/home/c0d3/src/esp-idf.git/components/spi_flash/include -I/home/c0d3/src/esp-idf.git/components/ulp/include -I/home/c0d3/src/esp-idf.git/components/vfs/include -I/home/c0d3/src/esp-idf.git/components/xtensa-debug-module/include -I/home/c0d3/src/esp-idf.git/components/wpa_supplicant/include -I/home/c0d3/src/esp-idf.git/components/wpa_supplicant/port/include -I/home/c0d3/src/esp-idf.git/components/ethernet/include -I/home/c0d3/src/esp-idf.git/components/app_trace/include -I/home/c0d3/src/esp-idf.git/components/app_update/include -I/home/c0d3/src/esp-idf.git/components/pthread/include -I/home/c0d3/src/esp-idf.git/components/smartconfig_ack/include -I/home/c0d3/src/esp-idf.git/components/sdmmc/include -DIDF_VER=\"v3.3-beta1-268-g5c88c5996\" -DFFCONF_H=\"lib/oofatfs/ffconf.h\" -DMICROPY_PY_BTREE=1 -c -MD -o build/network_ppp.o network_ppp.c
network_ppp.c:52:5: error: unknown type name 'ppp_pcb'
     ppp_pcb *pcb;
     ^~~~~~~
network_ppp.c:56:18: error: field 'pppif' has incomplete type
     struct netif pppif;
                  ^~~~~
network_ppp.c:61:27: error: unknown type name 'ppp_pcb'
 static void ppp_status_cb(ppp_pcb *pcb, int err_code, void *ctx) {
                           ^~~~~~~
network_ppp.c:96:34: error: unknown type name 'ppp_pcb'
 static u32_t ppp_output_callback(ppp_pcb *pcb, u8_t *data, u32_t len, void *ctx) {
                                  ^~~~~~~
network_ppp.c: In function 'pppos_client_task':
network_ppp.c:110:13: error: implicit declaration of function 'pppos_input_tcpip'; did you mean 'gpio_input_get'? [-Werror=implicit-function-declaration]
             pppos_input_tcpip(self->pcb, (u8_t*)buf, len);
             ^~~~~~~~~~~~~~~~~
             gpio_input_get
network_ppp.c: In function 'ppp_active':
network_ppp.c:127:25: error: implicit declaration of function 'pppapi_pppos_create' [-Werror=implicit-function-declaration]
             self->pcb = pppapi_pppos_create(&self->pppif, ppp_output_callback, ppp_status_cb, self);
                         ^~~~~~~~~~~~~~~~~~~
network_ppp.c:127:59: error: 'ppp_output_callback' undeclared (first use in this function); did you mean 'dns_found_callback'?
             self->pcb = pppapi_pppos_create(&self->pppif, ppp_output_callback, ppp_status_cb, self);
                                                           ^~~~~~~~~~~~~~~~~~~
                                                           dns_found_callback
network_ppp.c:127:59: note: each undeclared identifier is reported only once for each function it appears in
network_ppp.c:127:80: error: 'ppp_status_cb' undeclared (first use in this function); did you mean 'mp_state_ctx'?
             self->pcb = pppapi_pppos_create(&self->pppif, ppp_output_callback, ppp_status_cb, self);
                                                                                ^~~~~~~~~~~~~
                                                                                mp_state_ctx
network_ppp.c:132:13: error: implicit declaration of function 'pppapi_set_default' [-Werror=implicit-function-declaration]
             pppapi_set_default(self->pcb);
             ^~~~~~~~~~~~~~~~~~
network_ppp.c:133:13: error: implicit declaration of function 'pppapi_connect'; did you mean 'lwip_connect'? [-Werror=implicit-function-declaration]
             pppapi_connect(self->pcb, 0);
             ^~~~~~~~~~~~~~
             lwip_connect
network_ppp.c:143:13: error: implicit declaration of function 'pppapi_close'; did you mean 'lwip_close'? [-Werror=implicit-function-declaration]
             pppapi_close(self->pcb, 0);
             ^~~~~~~~~~~~
             lwip_close
network_ppp.c:157:13: error: implicit declaration of function 'pppapi_free'; did you mean 'mp_map_free'? [-Werror=implicit-function-declaration]
             pppapi_free(self->pcb);
             ^~~~~~~~~~~
             mp_map_free
network_ppp.c: In function 'ppp_ifconfig':
network_ppp.c:183:35: error: implicit declaration of function 'ppp_netif'; did you mean 'ppp_active'? [-Werror=implicit-function-declaration]
             struct netif *pppif = ppp_netif(self->pcb);
                                   ^~~~~~~~~
                                   ppp_active
network_ppp.c:183:35: error: initialization of 'struct netif *' from 'int' makes pointer from integer without a cast [-Werror=int-conversion]
network_ppp.c:185:59: error: dereferencing pointer to incomplete type 'struct netif'
                 netutils_format_ipv4_addr((uint8_t*)&pppif->ip_addr, NETUTILS_BIG),
                                                           ^~
cc1: all warnings being treated as errors
make[2]: *** [../../py/mkrules.mk:47: build/network_ppp.o] Error 1
make[2]: Leaving directory '/home/c0d3/src/micropython.git/ports/esp32'
make[1]: *** [Makefile:15: ports/esp32] Error 2
make[1]: Leaving directory '/home/c0d3/src/micropython.git'
make: *** [Makefile:12: esp32] Error 2
~~~

robert-hh, 2019-06-08:
After checking out 5c88c5996, did you repeat 
git submodule update --init
in the esp-idf?

c0d3z3r0, 2019-06-08:
Yep, but freertos is no submodule

robert-hh, 2019-06-09:
it looks like a problem with ./components/lwip/lwip/src/include/netif/ppp/ppp.h and ./components/lwip/lwip/src/include/netif/ppp/pppos.h. in the lwip module.

c0d3z3r0, 2019-06-09:
WTF. Cannot reproduce this anymore.... seems this was simply some PATH/env/whatever mistake..

closed 2019-06-09 by c0d3z3r0