micropython/micropython #4840: esp32: threading: stack check seems seems wrong

c0d3z3r0, 2019-06-10:
Hi there,

my application always raised an recursion exceed exception so I decided to do some test.
First some basic information:
- esp32 (esp-wroom-32)
- stack lenght increased to 50 (50*1024) for testing (same issues with the default of 16)
- master branch

This is my test code, a simple recursive function:

~~~py
def test(n):
    print(n)
    test(n+1)
~~~

Test 1: STACK_CHECK=0, no threading

~~~
>>> test(0)
0
...
157
1***ERROR*** A stack overflow in task mp_task has been detected.
abort() was called at PC 0x4008d5bc on core 1

Backtrace: 0x4008d2a7:0x3ffbd700 0x4008d5a5:0x3ffbd720 0x4008d5bc:0x3ffbd740 0x4008f3ea:0x3ffbd760 0x4008e558:0x3ffbd790 0x4008e50e:0x00000000

Rebooting...
~~~

Test 2: STACK_CHECK=1, no threading

~~~
>>> test(0)
0
...
153
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
...
  File "<stdin>", line 2, in test
RuntimeError: maximum recursion depth exceeded
~~~

Test 3: STACK_CHECK=0, threaded

~~~
>>> _thread.start_new_thread(test, (1,))
0
...
54
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4011431c  PS      : 0x00060330  A0      : 0x80118905  A1      : 0x3ffbb9d0  
A2      : 0x00000029  A3      : 0x3ffd1e28  A4      : 0x3ffb5188  A5      : 0x00000000  
A6      : 0x0000000a  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x3ffbb9b0  
A10     : 0x3ffbaaf8  A11     : 0x3ffbb9b0  A12     : 0x3ffb0fac  A13     : 0x3ffbb9a0  
A14     : 0x3ffb0fac  A15     : 0x2a0a4fd8  SAR     : 0x00000019  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000033  LBEG    : 0x40116045  LEND    : 0x4011604e  LCOUNT  : 0x00000000  

Backtrace: 0x4011431c:0x3ffbb9d0 0x40118902:0x3ffbb9f0 0x401189e1:0x3ffbba10 0x4010e0df:0x3ffbba40 0x4008e63d:0x3ffbba70

Rebooting...
~~~

Test 4: STACK_CHECK=1, threaded

~~~
>>> _thread.start_new_thread(test, (1,))
0
...
10
Unhandled exception in thread started by <function test at 0x3ffe4eb0>
Traceback (most recent call last):
  File "<stdin>", line 3, in test
...
  File "<stdin>", line 2, in test
RuntimeError: maximum recursion depth exceeded
~~~

Well, the first two tests look ok.  There's not much different in maximum stack size.

Where does this big difference in stack size (~150 vs. ~50) between threaded and unthreaded calls come from?

Test 4 looks like a bug, doesn't it? The maximum stack size should be about ~50.

c0d3z3r0, 2019-06-10:
Update: I increased the maximum stack size to 100. While I get 313 unthreaded recursions, threaded ones stay at 10!

closed 2019-06-10 by c0d3z3r0

dpgeorge, 2019-06-11:
You can set the default stack size for threads using `_thread.stack_size(...)` (same as CPython).

Note that (with STACK_CHECK=0) a crash of the system does not say much about maximum possible recursion depth because by the time the system crashes the stack may have overflowed long before that.

c0d3z3r0, 2019-06-11:
@dpgeorge that was what I tried before anything else but I was confused because it does not return the set value but the previous one... This is the same behaviour as in CPython but it is not documented very well there...
Thanks!