micropython/micropython #4841: [suggestion/question] A simple way to inject input / redirect output to stdio?

Alfrederson, 2019-06-11:
In the ESP32 version of NodeMCU (the lua interpreter) you can do this by calling a function to process an input string placed into a certain buffer and by 1) creating a function that accepts the same parameters as stdout and returns an integer 2) passing that function to esp_vfs_register, so that commands like printf write to that function instead of stdout, and it can call the old stdio, which writes to the uart interface

This way you can have a task running in FreeRTOS that takes data received via the built in bluetooth running in classic mode and issue commands to the interpreter itself, for example

How difficult would it be to do something similar within micropython?

dpgeorge, 2019-06-12:
Did you see `uos.dupterm()` ?  http://docs.micropython.org/en/latest/library/uos.html?highlight=dupterm#uos.dupterm

Alfrederson, 2019-06-12:
> Did you see `uos.dupterm()` ? http://docs.micropython.org/en/latest/library/uos.html?highlight=dupterm#uos.dupterm

That looks like something to be called from python, I'd need to expose the bluetooth functionality to python in order to use it, wouldn't I?

I'm looking at the uart interrupt handler and it looks like all it does is place the received data on a ring buffer, so I think doing what I want to micropython will probably be a lot easier than doing it to eLua. Redirecting stdout to bluetooth classic is pretty much trivial, so I'll give it a go

Alfrederson, 2019-06-14:
Well no luck getting micropython to build with bluetooth classic
There's no menuconfig to enable the component so that the toolchain will find the necessary files

dpgeorge, 2019-08-19:
`dupterm` and/or hooking into the `mp_hal_std*` C-level functions are the way to do this, so closing.

closed 2019-08-19 by dpgeorge