micropython/micropython #4847: NRF52832:Nus error nrf connect display ERROR(0X22) GATT CONN LMP TIMEOUT

SSEHX, 2019-06-15:
hi everyone
when i use nrf52832_xxaa run my micropython app have some error
MICROPY_PY_BLE_NUS enabled
use redme note5 nrf connect app connect my device , when send some data after ,can see some error : ERROR(0X85):GATT ERROR and ERROR:(0x22):GATT CONN LMP TIMEOUT
every 40 second connect lost 
this is device log
0INFO		:40 
�0INFO		:6E 
�0INFO		:

�0INFO		:ADV: uuid size: 16, type: 20, uuid: 1, vs_idx: 2

�0INFO		:Set Adv data size: 27

�0INFO		:GAP CONNECT evt id ： 10
�0INFO		:>>> unhandled evt: 0x23

�0INFO		:GATTS EVT EXCHANGE MTU REQUEST

�0INFO		:GATTS write

�0INFO		:>>> unhandled evt: 0x01                                  //can not use my phone send data

�0INFO		:GAP CONN PARAM UPDATE                             //have this event device is disconnect  

�0INFO		:GAP DISCONNECT

�0INFO		:Device name applied

�0INFO		:encoded uuid for service 0: 
�0INFO		:9E 
�0INFO		:CA 
�0INFO		:DC 
�0INFO		:24 
�0INFO		:0E 
�0INFO		:E5 
�0INFO		:A9 
�0INFO		:E0 
�0INFO		:93 
�0INFO		:F3 
�0INFO		:A3 
�0INFO		:B5 
�0INFO		:01 
�0INFO		:00 
�0INFO		:40 
�0INFO		:6E 
�0INFO		:

�0INFO		:ADV: uuid size: 16, type: 20, uuid: 1, vs_idx: 2

�0INFO		:Set Adv data size: 27

�0INFO		:GAP CONNECT evt id ： 10
�0INFO		:>>> unhandled evt: 0x23

�0INFO		:GATTS EVT EXCHANGE MTU REQUEST

�0INFO		:GATTS write

I really hope to get help, thank you very much.

glennrub, 2019-06-19:
@SSEHX, It sounds a bit similar to the issue #4288. Could you try adding a new switch-case in the `ble_drv.c` -> `ble_evt_handler()`function  to handle the`BLE_GAP_EVT_DATA_LENGTH_UPDATE_REQUEST`? In the referenced issue there are two proposals, could you see if any of those works for you?

SSEHX, 2019-06-28:
sorry ,i try it,but not work
Still no effect, the mobile terminal displays ERROR 34 (0X22) GATT CONN LMP TIMEOUT