micropython/micropython #4848: ESP32 SPI SD Card - Invalid DMA Channel

McSpider, 2019-06-15:
I'm having problems getting a SD card to work over SPI with the new MicroPython SD card support introduced in 8e3af7d & 6077d17 by @dpgeorge and @nickovs

The SD card and hardware work fine when using Arduino.

- Micropython version: `esp32-20190615-v1.11-45-g14cf91f70`
- Hardware: 
Wemos clone of a `MH-ET LIVE MiniKit` and a `D1 Mini SD Card Shield`
- SD Card connected as follows:  
`SCK: 18`  `MISO: 19`  `MOSI 23`  `CS: 5` 

The documentation gives this as an example: 
`uos.mount(storage.SDCard(), "/sd")` 
Which obviously doesn't work since SDCard is part of machine, fixing that results in:

    >>> uos.mount(machine.SDCard(), "/sd")
    E (23331) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
    Traceback (most recent call last):
      File "<stdin>", line 1, in <module>
    OSError: 16

So it defaults to MMC, which is fine, but the example should note that and perhaps provide a SPI example as well.

So then I tried:

    >>> import machine
    >>> uos.mount(machine.SDCard(slot=2), "/sd")
    E (15969) spi_master: spi_bus_initialize(235): invalid dma channel
    Traceback (most recent call last):
      File "<stdin>", line 1, in <module>
    OSError: 258

    >>> uos.mount(machine.SDCard(slot=2, sck=machine.Pin(18),miso=machine.Pin(19),mosi=machine.Pin(23),cs=machine.Pin(5)), "/sd")
    E (27309) spi_master: spi_bus_initialize(235): invalid dma channel
    Traceback (most recent call last):
      File "<stdin>", line 1, in <module>
    OSError: 258

So apparently the dma channel is wrong, but I can't change that anywhere in my code. And I don't know enough about the micropython code to start poking around in it. Not that I didn't try but that lead me to the espressif source code which made me even more clueless because IMO it should be working.

---

Some other things I tried:

    # This obviously won't work according to the docs, but I tried anyways
    >>> uos.mount(machine.SDCard(slot=1, sck=machine.Pin(18),miso=machine.Pin(19),mosi=machine.Pin(23),cs=machine.Pin(5)), "/sd")
    Traceback (most recent call last):
      File "<stdin>", line 1, in <module>
    OSError: 259

No idea what OSError 259 is, and I couldn't find any documentation on it, same for 258, but at least that tells me `invalid dma channel` above it.

    >>> uos.mount(machine.SDCard(sck=machine.Pin(18),miso=machine.Pin(19),mosi=machine.Pin(23),cs=machine.Pin(5)), "/sd")
    E (32129) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
    Traceback (most recent call last):
      File "<stdin>", line 1, in <module>
    OSError: 16

Doesn't work, we actually want to use SPI mode not MMC. Not sure if it's worth making the code detect this case?

nickovs, 2019-06-16:
Sorry about this. There is indeed a bug in the handling of SD cards on the SPI interfaces due to my failing to properly test the code after I changed the handling of the slot numbers at the last minute. I have submitted a (properly tested) PR #4851 to fix this. 

The PR also corrects the typo in the documentation stemming from moving the `SDCard` class into the `machine` module.

The code fix is a one-liner, so if you are building from source you can try it for yourself. Hopefully it will fix your issues.

McSpider, 2019-06-16:
That did indeed solve my issue, thanks!

dpgeorge, 2019-06-17:
Should be fixed by 1a51fc9ddf889c6a81e0156e6d438e4228479874

closed 2019-06-17 by dpgeorge

justinmoon, 2019-07-26:
I'm still getting this on an ESP32 m5stack:

```
>>> uos.mount(machine.SDCard(), "/sd")
E (23331) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 16
```

nickovs, 2019-07-26:
The m5stack has the SD card wired to non-standard pins. See the [documentation](https://docs.m5stack.com/#/en/core/basic?id=peripherals-pin-map) for details.

For the TL;DR, try using:
```
machine.SDCard(slot=2, mosi=23, miso=19, sck=18, cs=4)
```
I hope that helps.

justinmoon, 2019-07-26:
Thanks! That worked!!

johncblacker, 2020-06-23:
FWIW, I got my card working on an ESP32-CAM board with: uos.mount(slot=1,  mosi=23, miso=19, sck=18,  cs=5), "/sd")

cvidalmo, 2020-07-13:
Solution to access the SDCard of the ESP32 CAM AI Thinker board. (Solução para acessar o SDCard da placa ESP32 CAM AI Thinker).

import machine, sdcard, os
from machine import Pin, SPI
spisd = SPI(1, sck=Pin(14), mosi=Pin(15), miso=Pin(2))
sd = sdcard.SDCard(spisd, machine.Pin(13))
os.mount(sd, '/sd')
os.listdir("/sd")

Contact: Carlos Vidal (Brazil)
cvidalmo@gmail.com