micropython/micropython #4850: xtensa-esp32-elf/sys-include/stdlib.h:155:44

CXianRen, 2019-06-16:
xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdlib.h:155:44: error: expected initializer before '__result_use_check'
 void *reallocarray(void *, size_t, size_t) __result_use_check __alloc_size(2)

ESPIDF:5c88c5996dbde6208e3bec05abc21ff6cd822d26
and the version of xtensa-esp32-elf: https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_2_0-esp32-2019r1-linux-amd64.tar.gz
gcc version: gcc (Ubuntu 5.4.0-6ubuntu1~16.04.11) 5.4.0 20160609

i try all steps in Readme and other.
finally i change the version of xtensa-esp32-elf.
it work in version:https://dl.espressif.com/dl/xtensa-esp32-elf-linux64-1.22.0-61-gab8375a-5.2.0.tar.gz 
i don't know why. but it work. and i think it may cause by the gcc version.

dpgeorge, 2019-06-17:
The issue is that the currently-supported IDF commit is only compatible with the crosstool-ng-1.22.0-80-g6c4433a toolchain.

It's probably a good time now to update the esp32 port to a more recent IDF commit, that uses the latest toolchain.

sanu-krishnan, 2019-06-17:
Can we use any other versions than "xtensa-esp32-elf-linux64-1.22.0-80-g6c4433a-5.2.0" to resolve this for the time being ?

sanu-krishnan, 2019-06-17:
Hi @CXianRen, I tried your solution, but I am getting some errors here. 
```
Traceback (most recent call last):
  File "../../../esp-idf//tools/ldgen/ldgen.py", line 22, in <module>
    from fragments import FragmentFileModel
  File "/home/openmaker/ebot-iot/esp-idf/tools/ldgen/fragments.py", line 20, in <module>
    from sdkconfig import SDKConfig
  File "/home/openmaker/ebot-iot/esp-idf/tools/ldgen/sdkconfig.py", line 18, in <module>
    from pyparsing import Word, printables, Combine, Literal, hexnums, quotedString, Optional, nums, removeQuotes, oneOf, Group, infixNotation, opAssoc
ModuleNotFoundError: No module named 'pyparsing'
Makefile:808: recipe for target 'build/esp32.common.ld' failed
make: *** [build/esp32.common.ld] Error 1

```

dpgeorge, 2019-06-17:
> It's probably a good time now to update the esp32 port to a more recent IDF commit,

For that, see #4853 

> that uses the latest toolchain.

Unfortunately there is no "stable" branch/tag that uses the very latest toolchain.  MicroPython currently uses v3.x of the ESP IDF and that seems to require 1.22.0-80-g6c4433a-5.2.0 of the toolchain.  And at this stage I don't think it's a good idea to update to ESP IDF v4.0-dev because it requires a lot of changes (v3->v4) and v4 is still a work-in-progress.

CXianRen, 2019-06-17:
@sanu-krishnan  I think you are using python3,  you just need set PYTHON in makefile or install pyparsing in python2.x.

dpgeorge, 2019-06-19:
An update to the ESP IDF was made in 3ee3995be1c6c461110d7908498777241e08a76a.  The esp32 port requires toolchain 1.22.0-80-g6c4433a-5.2.0 to build.

closed 2019-06-19 by dpgeorge

projectgus, 2019-07-16:
Hi all! Was pointed to this thread by a [post on our forum](https://esp32.com/viewtopic.php?f=13&t=11366&p=46390#p46387) as there is some confusion about how to get the correct toolchain.

If you follow the "latest" ESP-IDF Get Started Guide, the toolchain is too new for MicroPython (at time of writing "latest" IDF is v4.0-dev, MicroPython is using v3.3-beta3, and v4.0 includes some non-backwards-compatible changes).

You can get the correct toolchain to build MicroPython for ESP32 by following either the ESP-IDF Get Started guide for the v3.3 "release branch", or the stable release version (stable version is v3.2.2 at time of writing):
https://docs.espressif.com/projects/esp-idf/en/stable/
https://docs.espressif.com/projects/esp-idf/en/release-v3.3

dpgeorge, 2019-07-16:
Thanks @projectgus .  As part of this issue I did update the esp32 README to point to the stable idf, see 3ee3995be1c6c461110d7908498777241e08a76a

projectgus, 2019-07-16:
@dpgeorge Awesome, thanks - sorry I didn't think to check the commit.