micropython/micropython #4854: Secure boot functionality for esp32

akisKING, 2019-06-17:
Has secure boot feature on ESP32 been implemented on micropython?. If not then what is the other method to secure the firmware?

hetvishah08, 2019-06-18:
I have the same question.

dpgeorge, 2019-06-18:
No it's not currently implemented.  There is currently no way to secure the firmware on esp32.

MrSurly, 2019-07-03:
@dpgeorge Would a PR implementing this be a welcome change? 

More broadly, we're looking to fully secure MP on the ESP32 (Also see https://github.com/micropython/micropython/pull/3576):

* Secure boot
* Encrypted partitions
* Encrypted filesystem partition
* Disabling CTRL-C / REPL
* Probably other stuff I'm forgetting

Ostensibly, this would be a compile-time config file option that you'd have to intentionally enable, and wouldn't affect a normal build.

dpgeorge, 2019-07-04:
> Would a PR implementing this be a welcome change?

Yes, but it would be best to do it a step at a time.  Probably it's worth getting #3576 done first and in a way that will extend to having encrypted partitions/boot/update.

> Disabling CTRL-C / REPL

That's something which could be useful for all ports.  Ctrl-C can already be disabled using `micropython.kbd_intr(-1)`, and/or disabling all dupterm sources.

MrSurly, 2019-07-24:
> That's something which could be useful for all ports. Ctrl-C can already be disabled using micropython.kbd_intr(-1), and/or disabling all dupterm sources.

Is there any timing considerations whereas if someone is quick on the keyboard could hit CTRL-C before this executes?  Maybe have an option to disable CTRL-C for `boot.py`, which could then execute this?

MrSurly, 2019-07-24:
>Yes, but it would be best to do it a step at a time. Probably it's worth getting #3576

@dpgeorge #3576 or #4910?

dpgeorge, 2019-07-25:
> Is there any timing considerations whereas if someone is quick on the keyboard could hit CTRL-C before this executes?

Right, that's a good point.  REPL is enabled before boot.py runs and ctrl-C is enabled during execution of it.  Need to think about how to deal with this.

> #3576 or #4910?

I want to go with #4910 because it has support for encrypted partitions (see also the other comments about it compared to #3576).  If you can test #4910 that would be great.

MrSurly, 2019-08-01:
>  If you can test #4910 that would be great.

I'll look into it.

git-n-pissed, 2023-01-23:
Any update on this?  #3576 was obsoleted by #4910, which has been worked.  @dpgeorge, is there any testing or other support that would help get this going again?