micropython/micropython #4856: Memory space that remains unaffected on reset on ESP32

hetvishah08, 2019-06-18:
Is there any way to store some variables which doesn't get affected on Reset?
 I saw this memory partition named as NVS in Arduino ESP32 and ESP-IDF. This might work for me but I guess it hasn't been implemented on Micropython yet. 
 Any other way to store variables that remain unchanged on reset?
I want to store password credentials after acquiring once into memory space that is unaffected by reset.

jimmo, 2019-06-18:
Many MicroPython ports (including ESP32) provide a filesystem backed by flash. You can read and write files using the standard Python file APIs (e.g. `open()`). You might want to use the `ujson` to encode your data, or sometimes it's just simpler to write it out as a Python file (e.g. `config.py`) that you can import directly (`import config` / `config.MY_VARIABLE`). (If you've used WebREPL, this is how it stores its config).

You can find more info about this in the [MicroPython Forum](https://forum.micropython.org).

hetvishah08, 2019-06-18:
As I mentioned earlier, I want to store passwords that shouldn't be accessible to anybody. By using your method I will be able to create a file on the system but it can be accessed via REPL, which I don't want for security reasons.
That's why I want a memory space where i can store the password once.

jimmo, 2019-06-18:
Does your MicroPython code needs to be able to access these passwords later? Whatever API exists to make this available means that if your code can access it, then so can the REPL (regardless of where they're stored).

One option would be to never provide this API to Python, and confine all access to this data a C module (i.e. the private data would never be accessible to a Python method). But this is starting to get very specific to a given application (e.g. let's say it's the auth keys for a remote server, it means the C code has to do the auth and connection). But if you were to go down this route, then you could certainly use NVS to implement it.

NVS in a sense works just like the MicroPython filesystem - it provides a convenient key/value API on top of spiflash.

rolandvs, 2019-06-18:
Probably the best thing is to add a secure element like the ATECC608 from Atmel/MCP via I2C...

hetvishah08, 2019-06-18:
@jimmo Yes it needs to access those passwords after reset or whenever called in main.py. I haved used ucryptolib as an alternate to encrypt data and make it difficult to understand even if accessed through REPL. But I was looking for a memory solution. 
NVS could be a solution but it hasn't been implemented yet?

hetvishah08, 2019-06-18:
@rolandvs currently I'm looking for a software solution. I will look into it for future updates in the hardware.

WayneKeenan, 2019-06-18:
This is how to persist data in RAM over a reset in C, so would still require MicroPython plumbing, but at least you know you can.

https://esp32.com/viewtopic.php?t=4931

jimmo, 2019-06-18:
> @jimmo Yes it needs to access those passwords after reset or whenever called in main.py. I haved used ucryptolib as an alternate to encrypt data and make it difficult to understand even if accessed [...]

I don't see how NVS solves your problem though. It's just another "filesystem" backed by the same flash. If your main.py can call a method like `nvs.get("key")`, then so can the REPL. You could use ucryptolib, but now you need somewhere to store your crypto keys, so you end up with the same problem. You just can't have secret data in your Python code.

However, there's one possible way you could address this while still only writing Python - that is to build a frozen module that contains:
 - your keys that you're using with ucryptolib
 - any code that needs to work with the secret data

As long as your frozen module doesn't have a method to access the key directly, but rather a `connect()` or `do_thing_with_secret_data()`, then it's much less trivial for someone accessing the REPL. (i.e. this is basically a way of doing what I suggested earlier but in Python instead of C).

This means that if someone can access the bytecode of your frozen module from the REPL then there's still a way in. The obvious way to do this is via machine.mem8. I'm not aware of a simpler way to do this, but I haven't looked in detail. But you can at least remove support for machine.mem8 from your firmware.

Note that someone can still physically connect to the flash chip and dump its contents. So flash encryption is important too.

> This is how to persist data in RAM over a reset in C, so would still require MicroPython plumbing, but at least you know you can.
> 
> https://esp32.com/viewtopic.php?t=4931

This is neat, but I would imagine @hetvishah08 wants this data to be persisted across power cycles? And ultimately it has the same limitation as using NVS -- if a Python API existed for it, then anyone (main.py, the REPL, etc) can access it.

> Probably the best thing is to add a secure element like the ATECC608 from Atmel/MCP via I2C...

@rolandvs  I'm not super familiar with the details of secure elements, but I don't see a way this addresses the "need to be able to access this from main.py but not the REPL" problem. Generally my understanding was that you still require some sort of non-persisted key (e.g. interactive, or challenge-response) in order to interact with the secure element.

WayneKeenan, 2019-06-18:
> And ultimately it has the same limitation as using NVS -- if a Python API existed for it, then anyone (main.py, the REPL, etc) can access it.

Not necessarily, It could be a write only call from MicroPython so only a custom C module could read it and use it.

hetvishah08, 2019-06-18:
> However, there's one possible way you could address this while still only writing Python - that is to build a frozen module that contains:
> 
> * your keys that you're using with ucryptolib
> * any code that needs to work with the secret data
> 
Thank you for your help @jimmo . I have already made a different module to encrypt and decrypt the password and other credentials. I will add this module into frozen modules and call it in main.py. 

> As long as your frozen module doesn't have a method to access the key directly, but rather a `connect()` or `do_thing_with_secret_data()`, then it's much less trivial for someone accessing the REPL. [...]
I don't know about machine.mem8. I'll look into it 

> Note that someone can still physically connect to the flash chip and dump its contents. So flash encryption is important too.
> 
I just want to know whether Flash Encryption has been implemented yet? Also I would like to know about Secure boot.

hetvishah08, 2019-06-18:
Thank you for your help @WayneKeenan. My concern is to retain data at any reset.

josei, 2020-08-02:
@hetvishah08 Did you make any progress with this? Regarding the proposal about frozen modules, I wonder how it's possible to hide the secret, as long as python has no way to declare private variables (actually, micropython 1.12 does not even perform name mangling of any kind). Therefore, the keys that you're using with ucryptolib would be visible to any other module using it. This could be fixed using constants, which are completely hidden to external modules, though they're only supported for integers.