micropython/micropython #4857: STM32 USE_HOST_MODE

IAmCoder, 2019-06-18:
If I define `USE_HOST_MODE` to compile in the code for experimental USB OTG support here:
https://github.com/micropython/micropython/blob/master/ports/stm32/usb.c#L815, I get a host of compile errors. First is that the includes seem to have been moved to sub-directories, which I can fix like this:

    #include "usbhost/Core/Inc/usbh_core.h"
    #include "usbhost/Core/Src/usbh_core.c"
    //#include "usbh_usr.h"
    #include "usbhost/Class/HID/Inc/usbh_hid_core.h"
    #include "usbhost/Class/HID/Inc/usbh_hid_keybd.h"
    #include "usbhost/Class/HID/Inc/usbh_hid_mouse.h"
    #include "usbhost/Core/Src/usbh_hcs.c"

Then there are a lot of missing references that I tried adding from the downloads from ST.com `stm32_f105-07_f2_f4_usb-host-device_lib` (STSW-STM32046 https://www.st.com/en/embedded-software/stsw-stm32046.html) and `STM32Cube_FW_H7_V1.4.0` (STM32CubeH7) under `/usbhost/Class/HID/`, `/usbhost/Class/OTG/` and `/usbhost/Core/`, but that causes many conflicts with the files under `/usbdev/`.

Could someone who managed to get the experimental USB OTG support going with a mouse or keyboard point me in the right direction?

dpgeorge, 2019-06-18:
Sorry but the USB host support in the code is really outdated and needs a bit of work to get it running again.

jonnor, 2025-10-03:
Is this still a problem?

dpgeorge, 2025-10-04:
Yes, it's still broken.