micropython/micropython #4861: Data sent and received are inconsistent about REPL on UART6  pyb1.0  STM32F405

YSP0316, 2019-06-21:
HELP! HELP! HELP!

The data returned by the repl is lost compared to the data received and will not be executed correctly；
The part of the picture marked in red is abnormal data, which often occurs！
References are pyb1.0 firmware.
I hope someone can help solve this problem.
[image]

peterhinch, 2019-06-21:
You are experiencing occasional missed characters. This is probably because you're sending data at a rate too high for the Pyboard's UART driver to process it. Options are to use hardware flow control or to try a lower baud rate.

dpgeorge, 2019-09-27:
Closing due to inactivity.  If it's still a problem and you can provide reproducible tests, feel free to reopen.

closed 2019-09-27 by dpgeorge