micropython/micropython #4864: Trying (to the best of my ability) to install MicroPython in Ubuntu 19.04

baqwas, 2019-06-22:
Running under (with sudo privileges):
```
matha@myUbuntu:~/projects/micropython/ports/unix$ uname -a
Linux BeUlta 5.0.0-17-generic #18-Ubuntu SMP Tue Jun 4 15:34:08 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
```
Latest update/upgrade applied.

What worked: :)
```
sudo apt install build-essential libreadline-dev libffi-dev git pkg-config
cd ~/projects # just a personal modus operandi; not mandatory
git clone --recurse-submodules https://github.com/micropython/micropython.git
cd ./micropython/ports/unix
```
What doesn't work: (
```
matha@myUbuntu:~/projects/micropython/ports/unix$ make axtls
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
make: Nothing to be done for 'axtls'.
matha@myUbuntu:~/projects/micropython/ports/unix$ make axtls V=1
make: Nothing to be done for 'axtls'.
```
**How to move forward** at this juncture with the installation of MicroPython on this platform, please?

Thanks.

dlech, 2019-06-22:
If it says "Nothing to be done..." then it means that you have already compiled it and you can continue with the next step.

dpgeorge, 2019-06-24:
You shouldn't need to run `make axtls`, just `make` is enough for the unix port (as README.md describes).

baqwas, 2020-06-11:
Sorry for the neglect. Thx for closing it.
Kind regards.