micropython/micropython #4865: mp_obj_equal() returns a boolean even when .binary_op returns something else

benob, 2019-06-23:
I am implementing an array type with comparators which return arrays with the result of the comparison (like in numpy). It works for operators such as > or < but not == and !=.

In mp_binary_op(), MP_BINARY_OP_EQUAL and MP_BINARY_OP_NOT_EQUAL are handled separately by calling mp_obj_equal() which returns a boolean, while for other operators the result is that of the .binary_op method of the type.

I didn't check what CPython does, but those operators work like intended in numpy, and it would make sense that the handling be uniform for all binary operators.

dpgeorge, 2019-06-23:
Thanks for the report about this issue.  A test that shows the difference between CPython and MicroPython for this case is:
```python
class A:
    def __eq__(self, other):
        print('A __eq__')
        return [1,2]
print(A() == A())
print(A() == 1)
print(1 == A())
```
Output for CPython:
```
A __eq__
[1, 2]
A __eq__
[1, 2]
A __eq__
[1, 2]
```
Output for MicroPython:
```
A __eq__
False
A __eq__
False
A __eq__
False
```

dpgeorge, 2019-06-26:
A related issue: if a user-defined class has `__eq__(a, b)` returning `False` when `a is b` (ie same instance) then MicroPython gets it wrong and returns `True` regardless.

andrewleech, 2019-07-04:
On a possibly related note, I'm running into an inconsistency with cpython with regard to comparisons in user-defined objects to string; `__eq__` is not called in mpy. Perhaps it's getting caught by the same shortcut as `a is b`

``` python
class Test:
    def __init__(self):
        self.name = "test"
        self.value = (1,)
    def __eq__(self, other):
        print("in __eq__")
        return other == self.name or other == self.value
    #def __hash__(self):
    #    print("in __hash__")
    #    return hash(self.name)
    
print(Test() == (1, ))
print(Test() == "test")
```

cpython
``` 
in __eq__
True
in __eq__
True
```

micropython:
```
in __eq__
True
False
```

dpgeorge, 2019-07-04:
> I'm running into an inconsistency with cpython with regard to comparisons in user-defined objects to string; `__eq__` is not called in mpy

Yes, there's an additional shortcut/special case for string equality.