micropython/micropython #4866: dupterm readinto() method not called

lerouxb, 2019-06-24:
I have a class that inherits from IOBase. It has readinto() and write methods(). I passed an instance to os.dupterm(). My write method gets called and I can duplicate the terminal output on a lcd screen, but my readinto() never executes. Am I missing something obvious? Reading the commit message here, it should work: https://github.com/micropython/micropython/commit/035906419d925bf723dba21f9fbb22ca1c8021f1

dpgeorge, 2019-06-24:
What port/board is the code running on?  Which MicroPython version?

lerouxb, 2019-06-24:
port: esp32

Here's my minimal test case:

```python
import uio
import uos

class Screen(uio.IOBase):
    def __init__(self):
        super().__init__()
        # more stuff here
        self.reads = 0
        self.writes = 0

    def readinto(self, buf):
        self.reads += 1

    def write(self, buf):
        self.writes += 1

screen = Screen()
uos.dupterm(screen)

# later:
print(screen.reads) # always 0
print(screen.writes) # always > 0
```

The last commit of my micropython copy is 434975defac12125e550d42eb71d5454e0386d51 which is admittedly quite old. I read through lots of commits, issues, pull requests and it looks like nothing relevant changed.. I better update to the latest, try again and report back.

dpgeorge, 2019-06-24:
The esp32 doesn't automatically read from dupterm objects.  Instead your code must call `uos.dupterm_notify()` when there are new characters available to read.

lerouxb, 2019-06-24:
Thanks! I'll PR the docs as I can't find that mentioned anywhere.

closed 2019-06-24 by lerouxb