micropython/micropython #4869: Add I2S Driver to ESP32 Makefile

nevercast, 2019-06-24:
Related to #4170 and #4821 regarding I2S support.

I am implementing a CModule as a demonstration on how to write CModules for MicroPython and what useful things can be done with them, CModules:https://docs.micropython.org/en/latest/develop/cmodules.html

I wanted to demonstrate a Python library that would play predefined audio files baked in to the firmware, for this I would use I2S and the internal DAC of the ESP32. Implement it in a CModule and show it off with some Python wrapper code.

While I have had good success running a custom python script from the `micropython.mk` file to generate the audio content, and have written a module that will play back these files. I've hit a snag with the `i2s` driver missing from MicroPython.

The errors during compile look like this
```
audio_ctrl.c:(.text.audioctrl_init_driver+0x33): undefined reference to `i2s_driver_install'
audio_ctrl.c:(.text.audioctrl_init_driver+0x5c): undefined reference to `i2s_set_pin'
audio_ctrl.c:(.text.audioctrl_init_driver+0x64): undefined reference to `i2s_set_dac_mode'
audio_ctrl.c:(.text.audioctrl_init_driver+0x6c): undefined reference to `i2s_zero_dma_buffer'
```

and they are caused by `i2s.o` being missing from `ESPIDF_DRIVER_O` in `ports/esp32/Makefile`

https://github.com/micropython/micropython/blob/3ee3995be1c6c461110d7908498777241e08a76a/ports/esp32/Makefile#L259-L273

I would like to petition that this driver is added to Micropython today, preemptive of the other issues currently open regarding I2S support, and to be permissive of mine CModules example.

Perhaps we can also discuss if there is a cleaner way to implement the functionality of including other drivers with the CModules `micropython.mk` Makefile, instead of editing the port.

nevercast, 2019-06-24:
If I get a thumbs up from a MicroPython contributor that a Pull Request would be welcome, I'm more than happy to make the 1 line change and PR it.

dpgeorge, 2019-06-24:
> I would like to petition that this driver is added to Micropython today

Sounds good to me.  The code will anyway only be linked in if it's used.

> Perhaps we can also discuss if there is a cleaner way to implement the functionality of including other drivers with the CModules `micropython.mk` Makefile, instead of editing the port.

Usually you could just include a C file in micropython.mk, but for the esp32 it's a bit different due to the way it must put .o files in a .a file before linking.  So it really needs to be done here.

One improvement could be to automatically include all .c source files in a given IDF directory (eg `components/driver/*.c`) in the build using wildcards (or similar).  Feel free to look into this.

dpgeorge, 2019-06-25:
> Do I need to sign a CLA?

No.  But please do use a real email address for the commit author, and read the very last section of the top-level README.md (re licensing).

nevercast, 2019-06-25:
> No. But please do use a real email address for the commit author, and read the very last section of the top-level README.md (re licensing).

Seems if my email address is in my Github account it replaces my real name. I'll see if I can change that in Github settings.

dpgeorge, 2019-06-25:
Fixed by 2f262d5f9a8a77ec611548162100d6df8e9703c0

commit 2f262d5f (2019-06-25): esp32/Makefile: Include all driver/*.c source files in the build.

Fixes #4869.
closed 2019-06-25 by dpgeorge