micropython/micropython #4873: time.time() always returning 0

fragmuffin, 2019-06-25:
We're running a Pyboard-D SF6W and the RTC seems to hang from time to time.

```
MicroPython v1.10-445-ga9b1d3ca3-dirty on 2019-06-18; PYBD_SF6W with STM32F767IIK
Type "help()" for more information.
>>> import time
>>> time.time()
0
```

### Soft Reset (doesn't work)

Software accessible resets do not fix the issue... the following attempts to reboot are done without removing power.

**`machine.reset`**
```
>>> import machine
>>> machine.reset()
...
MicroPython v1.10-445-ga9b1d3ca3-dirty on 2019-06-18; PYBD_SF6W with STM32F767IIK
Type "help()" for more information.
>>> import time
>>> time.time()
0
```

**`pyb.hard_reset()`**
```
>>> import pyb
>>> pyb.hard_reset()
...
MicroPython v1.10-445-ga9b1d3ca3-dirty on 2019-06-18; PYBD_SF6W with STM32F767IIK
Type "help()" for more information.
>>> import time
>>> time.time()
0
```

### Physical Reset (ie: completely remove power)

However, if pyboard is completely powered off (tested for ~3sec).
The problem is resolved.

```
MicroPython v1.10-445-ga9b1d3ca3-dirty on 2019-06-18; PYBD_SF6W with STM32F767IIK
Type "help()" for more information.
>>> import time
>>> time.time()
473385659
```

fragmuffin, 2019-06-25:
Hi @dpgeorge : Peter Boin from Allume Energy here...
I believe Ethan came to you about this before I started with Allume.

I couldn't find a corresponding issue, but we believe you found root cause and made adjustments to fix... I apologise if I've created a duplicate.

...

Actually... looking at the timing, it seems that fix would have gone into `v1.11`...
I'll update our firmware and I'll post back here if we experience it again.

dpgeorge, 2019-06-25:
> MicroPython v1.10-445-ga9b1d3ca3-dirty

This is actually v1.11 on the stm32-pybd branch, so should have the RTC fix.  The relevant commit for the fix is c2886868b9ccbe8580f57135bb29779a5d36c1e8

fragmuffin, 2019-06-25:
In that case, since I don't have any advice on how to replicate the problem, is there anything you'd like me to check next time it occurs?

dpgeorge, 2019-06-25:
>  is there anything you'd like me to check next time it occurs?

Please run (without any reset):
```python
import pyb, stm
print(hex(pyb.RTC().info()))
for i in range(0, 0x94, 4):
    print('RCC[%02x] = %08x' % (i, stm.mem32[stm.RCC + i] & 0xffffffff))
for i in range(0, 0x4c, 4):
    print('RTC[%02x] = %08x' % (i, stm.mem32[stm.RTC + i] & 0xffffffff))
```

fragmuffin, 2019-06-27:
Hi @dpgeorge , it's happened again, running the lines you sent me...

```
>>> import pyb, stm
>>> print(hex(pyb.RTC().info()))
0x40003
>>> for i in range(0, 0x94, 4):
...     print('RCC[%02x] = %08x' % (i, stm.mem32[stm.RCC + i] & 0xffffffff))
... 
RCC[00] = 03038283
RCC[04] = 16404819
RCC[08] = 0000940a
RCC[0c] = 00000000
RCC[10] = 00000000
RCC[14] = 00000000
RCC[18] = 00000000
RCC[1c] = 00000000
RCC[20] = 00000000
RCC[24] = 00000000
RCC[28] = 00000000
RCC[2c] = 00000000
RCC[30] = 2010017f
RCC[34] = 00000000
RCC[38] = 00000002
RCC[3c] = 00000000
RCC[40] = 10000400
RCC[44] = 00000000
RCC[48] = 00000000
RCC[4c] = 00000000
RCC[50] = 3ef7b7ff
RCC[54] = 000000f3
RCC[58] = 00000003
RCC[5c] = 00000000
RCC[60] = ffffefff
RCC[64] = 6cf77fb3
RCC[68] = 00000000
RCC[6c] = 00000000
RCC[70] = 00008103
RCC[74] = 00000000
RCC[78] = 00000000
RCC[7c] = 00000000
RCC[80] = 00000000
RCC[84] = 24003000
RCC[88] = 24003000
RCC[8c] = 00000000
RCC[90] = 00000000
>>> for i in range(0, 0x4c, 4):
...     print('RTC[%02x] = %08x' % (i, stm.mem32[stm.RTC + i] & 0xffffffff))
... 
RTC[00] = 00000000
RTC[04] = 00002101
RTC[08] = 00000000
RTC[0c] = 00000017
RTC[10] = 00007fff
RTC[14] = 0000ffff
RTC[18] = 00000000
RTC[1c] = 00000000
RTC[20] = 00000000
RTC[24] = 00000000
RTC[28] = 00000000
RTC[2c] = 00000000
RTC[30] = 00000000
RTC[34] = 00000000
RTC[38] = 00000000
RTC[3c] = 00000000
RTC[40] = 00000000
RTC[44] = 00000000
RTC[48] = 00000000
```

To clarify, that's on a fresh build of [`master`](205c6d0dc94ef1ca11d7bcef2090ef49c2015e48) (as described above)

dpgeorge, 2019-06-28:
Thanks for the register dump.

> print(hex(pyb.RTC().info()))
> 0x40003

This means when the board started up the RTC was already running and enabled, clocked by the LSE (in non-bypass mode).

> RCC[70] = 00008103
> RCC[74] = 00000000

This confirms the above `0x40003` value.

Some further questions:
- What is the output of `machine.reset_cause()`?
- Does this happen after a wake from lightsleep/deepsleep, or from updating the firmware?
- Does this happen on just one particular board, or many?

fragmuffin, 2019-07-01:
> What is the output of `machine.reset_cause()`?

```python
>>> machine.reset_cause() == machine.HARD_RESET
True
```

> Does this happen after a wake from lightsleep/deepsleep, or from updating the firmware?

I haven't updated firmware much, but i don't recall it occurring immediately after an update.
Most of the time I'm synchronising our code-base to flash and resetting:

* mount flash
* run `rsync`
* unmount flash
* `pyb.hard_reset()`... done from the host with...

```python
comport.write(b'\x03\r\nimport pyb\r\npyb.hard_reset()\r\n')
```

> Does this happen on just one particular board, or many?

All the above feedback has come from a single board, but we've observed it on 2 separate boards.
I'll swap out this board for another, and flash the same firmware onto it to see if it's replicated on another.

dpgeorge, 2019-07-01:
Can you please run that code to dump info on a board that is known to have the issue, but when it is running correctly (ie time() returns a value that is changing)?

fragmuffin, 2019-07-01:
> Can you please run that code to dump info on a board that is known to have the issue, but when it is running correctly

no problem...
this is from the same board as all previous test runs.

```python
>>> import time
>>> time.time()
615257572
>>> import pyb, stm
>>> print(hex(pyb.RTC().info()))                                                   
0x80100                                                                            
>>> for i in range(0, 0x94, 4):
...     print('RCC[%02x] = %08x' % (i, stm.mem32[stm.RCC + i] & 0xffffffff))
...
RCC[00] = 03038283
RCC[04] = 16404819
RCC[08] = 0000940a
RCC[0c] = 00000000
RCC[10] = 00000000
RCC[14] = 00000000
RCC[18] = 00000000
RCC[1c] = 00000000
RCC[20] = 00000000
RCC[24] = 00000000
RCC[28] = 00000000
RCC[2c] = 00000000
RCC[30] = 201001ff
RCC[34] = 00000040
RCC[38] = 00000002
RCC[3c] = 00000000
RCC[40] = 10008404
RCC[44] = 00005000
RCC[48] = 00000000
RCC[4c] = 00000000
RCC[50] = 3ef7b7ff
RCC[54] = 000000f3
RCC[58] = 00000003
RCC[5c] = 00000000
RCC[60] = ffffefff
RCC[64] = 6cf77fb3
RCC[68] = 00000000
RCC[6c] = 00000000
RCC[70] = 00008200
RCC[74] = 00000003
RCC[78] = 00000000
RCC[7c] = 00000000
RCC[80] = 00000000
RCC[84] = 24003000
RCC[88] = 24003000
RCC[8c] = 00000000
RCC[90] = 00000000
>>> for i in range(0, 0x4c, 4):
...     print('RTC[%02x] = %08x' % (i, stm.mem32[stm.RTC + i] & 0xffffffff))
... 
RTC[00] = 00005252
RTC[04] = 00192701
RTC[08] = 00040000
RTC[0c] = 00000037
RTC[10] = 00007fff
RTC[14] = 0000ffff
RTC[18] = 00000000
RTC[1c] = 00000000
RTC[20] = 00000000
RTC[24] = 00000000
RTC[28] = 00004360
RTC[2c] = 00000000
RTC[30] = 00000000
RTC[34] = 00000000
RTC[38] = 00000000
RTC[3c] = 00000000
RTC[40] = 00000000
RTC[44] = 00000000
RTC[48] = 00000000
```

fragmuffin, 2019-07-01:
When I do a hard reset (with my sync script) I run...

```python
comport.write(b'\x03\r\nimport pyb\r\npyb.hard_reset()\r\n')
comport.close()  # immediately afterward
```

Do you think this could have an adverse effect on the RTC in its next cycle?

dpgeorge, 2019-07-15:
@fragmuffin I couldn't reproduce this issue at all.  Do you have any way it can be reproduced?

> RCC[70] = 00008200

This value points to a possible problem because it means the board is using LSI (internal 32k) not LSE (external oscillator).  Could you check this memory location (`hex(stm.mem32[stm.RCC+stm.RCC_BDCR])`) on all boards you have?  It should be 0x8107 after `time.time()` has executed at least once (regardless of whether the board was just powered up, hard reset, etc).

fragmuffin, 2019-08-16:
@dpgeorge : I can't find a reliable way to replicate the problem, but when it's occurring it's easy to replicate on resets.
After hard resetting, or re-plugging it in (to the parent PCB), it often fixes it, and doesn't occur again for a few days... other experiences have shown the issue turning up every 5min.

Regarding the oscillator selection:
Of the 4 boards I have...

**pyboards with serial number `335633703138`, `336233493138`**
```
MicroPython v1.10-445-ga9b1d3ca3-dirty on 2019-06-18; PYBD_SF6W with STM32F767IIK
Type "help()" for more information.
>>> import stm, time
>>> hex(stm.mem32[stm.RCC+stm.RCC_BDCR])
'0x7'
>>> time.time()
473385600
>>> hex(stm.mem32[stm.RCC+stm.RCC_BDCR])
'0x8107'
```

**pyboard with serial number `336233723138`**
```
MicroPython v1.10-445-ga9b1d3ca3-dirty on 2019-06-18; PYBD_SF6W with STM32F767IIK
Type "help()" for more information.
>>> import stm, time
>>> hex(stm.mem32[stm.RCC+stm.RCC_BDCR])
'0x8200'
>>> time.time()
473385739
>>> hex(stm.mem32[stm.RCC+stm.RCC_BDCR])
'0x8200'
```

**pyboard with serial number `335333563138`**
has been re-flashed with `v1.11` (built locally from 205c6d0dc94ef1ca11d7bcef2090ef49c2015e48)

This board was in a state that was repeatedly exhibiting the issue: `time.time() == 0`.
After a soft reset, the problem wasn't present.
Then immediately after a hard reset (and re-connecting to REPL)

```python
MicroPython v1.11 on 2019-06-25; PYBD-SF6W with STM32F767IIK
Type "help()" for more information.
>>> import stm, time
>>> hex(stm.mem32[stm.RCC+stm.RCC_BDCR])
'0x8103'
>>> time.time()
473385605
>>> hex(stm.mem32[stm.RCC+stm.RCC_BDCR])
'0x8103'

# then after a reset
>>> import pyb
>>> pyb.hard_reset()

>>> import time
>>> import stm, time
>>> hex(stm.mem32[stm.RCC+stm.RCC_BDCR])
'0x8103'
>>> time.time()
0
>>> hex(stm.mem32[stm.RCC+stm.RCC_BDCR])
'0x8103'
```

`v1.11` :-1: ?: The above results appear to conclude this, but they're also circumstantial.
I've unplugged and plugged into our parent PCB a few times now (in performing the above tests, to see if our interface is causing the issue)... and now I cannot replicate the above fault condition (on `335333563138`).
I've also hard-reset the parent board ~20 times, testing `time.time()` each time... and still cannot replicate.

Could this fault be triggered on a bad connection to another PCB?

**Our Work-around:**
Because we're using this to timestamp data transmitted to an external source, we're using the timestamp on the initial handshake, and aligning that with `time.ticks_ms()` as a sort of "soft RTC".

Please let me know if there's anything else you'd like me to test.
My apologies for the delay in response... we've been busy (in a good way) :+1: