micropython/micropython #4882: minimla port dosent work on pyboardV1.1

josephcpro, 2019-06-28:
I built and flashed the minimal port to the pyboardV1.1 and I don't see any serial device to connect to when I'm done flashing. In the main I see that it initializes uart but nothing shows up. What could I be doing wrong?

dpgeorge, 2019-06-28:
> I don't see any serial device to connect to when I'm done flashing. In the main I see that it initializes uart but nothing shows up

It puts a UART on pins X1/X2.  You need to connect something to that (eg USB-UART adaptor) to see the REPL.

josephcpro, 2019-06-28:
yeah man I did that and I still see nothing

josephcpro, 2019-06-28:
1.

dpgeorge, 2019-07-01:
I tested the minimal port on PYBv1.1 and it does give an initial prompt over UART (9600 baud on X9/X10), but entering anything leads to a fault.

dpgeorge, 2019-07-01:
The fault is fixed by 999733b1fb08fb07a31846115998dbf43a913327

closed 2019-07-01 by dpgeorge