micropython/micropython #4883: esp32 How to set priority of Interruption with micropython

HotIce0, 2019-06-28:
Please help! thx

dpgeorge, 2019-07-02:
It's currently not possible to do this in MicroPython on the esp32.  In fact, I'm not even sure the ESP IDF allows to configure timer ISR priority.

HotIce0, 2019-07-09:
> It's currently not possible to do this in MicroPython on the esp32. In fact, I'm not even sure the ESP IDF allows to configure timer ISR priority.

Yes, ESP IDF not allows to configure timer ISR priority. as I knew the four timer of ESP32 have default priority according to ESP32 official Document. 
`class machine.Timer(id, ...)`
I think the `id` is the specified the timer. So, I try to set two timer have different id. 0 and 1 . 
However, it is different from my expectations.

dpgeorge, 2019-07-10:
> So, I try to set two timer have different id. 0 and 1 .
> However, it is different from my expectations.

Different timer id's should work.  If there's a discrepancy between the docs and the code behaviour please describe in detail what you see.

Otherwise I think this issue can be closed because the original issue (timer ISR priority) is not possible on the esp32.

closed 2019-07-11 by HotIce0