micropython/micropython #4885: ESP32: Cannot connect to WebREPL over websocket

spierepf, 2019-06-30:
After downloading and flashing my Lolin D32 Pro with the (latest) SPIRAM version of the micropython firmware from: http://micropython.org/download I cannot connect to my device through webrepl. On the serial console I get:
`
MicroPython v1.11-64-gced340d73 on 2019-06-27; ESP32 module with ESP32
Type "help()" for more information.
>>> 
WebREPL connection from: ('192.168.0.9', 55600)
assertion "self->buf[0] & 0x80" failed: file "../../extmod/moduwebsocket.c", line 97, function: websocket_read
abort() was called at PC 0x4014f3b3 on core 0

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x4008cda3:0x3ffc1d40 0x4008d0f9:0x3ffc1d60 0x4014f3b3:0x3ffc1d80 0x400ef7f9:0x3ffc1db0 0x400efb33:0x3ffc1df0 0x400f16a2:0x3ffc1e30 0x400f1caa:0x3ffc1eb0 0x400e04e1:0x3ffc1ed0 0x400dc5a9:0x3ffc1ef0 0x400dc5ed:0x3ffc1f10 0x400dd4e8:0x3ffc1f40 0x400f4535:0x3ffc1fc0 0x400f190c:0x3ffc1ff0 0x400f997d:0x3ffc2010 0x400fa034:0x3ffc2030 0x400f1818:0x3ffc2070 0x400919b1:0x3ffc20a0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4928
ho 0 tail 12 room 4
load:0x40078000,len:10264
ho 0 tail 12 room 4
load:0x40080400,len:6740
entry 0x400806ec
I (510) spiram: Found 64MBit SPI RAM device
I (510) spiram: SPI RAM mode: flash 40m sram 40m
I (510) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (516) cpu_start: Pro cpu up.
I (520) cpu_start: Application information:
I (525) cpu_start: Compile time:     Jun 27 2019 12:35:17
I (531) cpu_start: ELF file SHA256:  0000000000000000...
I (537) cpu_start: ESP-IDF:          v3.3-beta1-694-g6b3da6b18
I (543) cpu_start: Starting app cpu, entry point is 0x40083014
I (534) cpu_start: App cpu up.
I (1419) spiram: SPI SRAM memory test OK
I (1420) heap_init: Initializing. RAM available for dynamic allocation:
I (1420) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1426) heap_init: At 3FFBA098 len 00025F68 (151 KiB): DRAM
I (1432) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1439) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1445) heap_init: At 40096438 len 00009BC8 (38 KiB): IRAM
I (1451) cpu_start: Pro cpu start user code
I (128) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
MicroPython v1.11-64-gced340d73 on 2019-06-27; ESP32 module with ESP32
Type "help()" for more information.
>>> 
`

dpgeorge, 2019-07-03:
Using the latest SPIRAM firmware on a LoLin32 Pro, webrepl works ok for me:
```
MicroPython v1.11-97-gd821a27b5 on 2019-07-03; ESP32 module with ESP32
>>> micropython.mem_info()
stack: 736 out of 15360
GC: total: 4098240, used: 6176, free: 4092064
 No. of 1-blocks: 38, 2-blocks: 12, max blk sz: 264, max free sz: 255696
>>> import webrepl
I (16279) modsocket: Initializing
>>> webrepl.start(password='mp')
WebREPL daemon started on ws://192.168.1.10:8266
Started webrepl in manual override mode
>>> 
WebREPL connection from: ('192.168.1.7', 42898)
(works as expected from here)
```
I use Firefox 67.0.1, 64-bit version on Linux, running current http://micropython.org/webrepl

spierepf, 2019-07-03:
I was connecting from http_s_://micropython.org/webrepl

PEBCAK

Thanks!

closed 2019-07-03 by spierepf