micropython/micropython #4888: Oled ssd 1306 driver 128*64 : OSError: [Errno 19] ENODEV

hwt9520, 2019-06-30:
from machine import I2C, Pin
i2c=I2C(scl=Pin(5),sda=Pin(4))
from ssd1306 import SSD1306_I2C

#all the code above is ok but:

oled = SSD1306_I2C(128, 64, i2c)

#give me error like this:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "ssd1306.py", line 100, in __init__
  File "ssd1306.py", line 36, in __init__
  File "ssd1306.py", line 61, in init_display
  File "ssd1306.py", line 105, in write_cmd
OSError: [Errno 19] ENODEV

#I use the mp version and the driver like this:
esp8266-20190630-v1.11-72-g9cebead27
https://github.com/micropython/micropython/blob/master/drivers/display/ssd1306.py

jimmo, 2019-06-30:
This means that the oled device cannot be found on the i2c bus at the default address. What does `i2c.scan()` return? Do you have pull-up resistors on the sda/scl lines?

You might be able to get more help on the forum. https://forum.micropython.org/

hwt9520, 2019-07-06:
I have searched keywords in forum like this: '1306 oled ENODEV', nothing got

I tested my CO2 senser in I2C mode:

from machine import I2C, Pin
i2c=I2C(scl=Pin(5),sda=Pin(4),freq=100000)
i2c.scan()
[81]
i2c.readfrom(81,9) 
b"\x01\xc2\x00\x00\x01'\xc6\x00\xdc"
so, everything is good.

return to oled,
from machine import I2C, Pin
i2c=I2C(scl=Pin(5),sda=Pin(4))
from ssd1306 import SSD1306_I2C
oled = SSD1306_I2C(128, 64, i2c)

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "ssd1306.py", line 100, in __init__
  File "ssd1306.py", line 36, in __init__
  File "ssd1306.py", line 61, in init_display
  File "ssd1306.py", line 105, in write_cmd
OSError: [Errno 19] ENODEV

i2c.scan()
[]

so, I think, either the driver or the oled have problem
but the oled is a new one, I just buy it last week

jimmo, 2019-07-06:
In that case can you start a new thread on the forum? Given that it's failing to respond to a scan, it doesn't look like a problem with the driver.

robert-hh, 2019-07-07:
Scan reports 81 dec as address. The driver's default is 60 dec (or 3c hex). So you have to specify the address as:
oled = SSD1306_I2C(128, 64, i2c, addr=81)

junhuanchen, 2019-07-07:
see this

i2c.scan()
[]

check your hardware connect.

find the i2c address of @robert-hh 's talk.

peterhinch, 2019-07-07:
@robert-hh The value of 81 was returned by the CO2 sensor, not the display. The display did not respond to the scan. So I think either:
- The display is broken.
- It is wired incorrectly.
- Or there is a problem with pullups.

mcauser, 2019-07-08:
Can you post a photo of the display module. Specifically the pins.
The SSD1306 display can be operated in I2C and SPI modes, depending on some onboard resistors.

If it's got 4 pins, it's almost certainly an I2C flavoured display.
5, 6 or 7 pins and it's most likely SPI.

Sometimes the display manufacturers label the pins SDA, SCL, making it seem like it should be I2C, but the SDA is actually a MISO and the SCL is actually a SCK.

Tapping the clock and data pins with a logic analyser may quickly reveal what's going on.

Koraze, 2019-10-01:
Hi,

I got some kind of the same issue ... I got two Lora32 chips with OLED screen (SSD1306 display) :
 - A Wemos one : https://alexnld.com/product/2pcs-wemos-ttgo-lora32-868-915mhz-esp32-lora-oled-0-96-inch-blue-display-bluetooth-wifi-esp-32-development-board-module-with-antenna/
 - A Heltec one : https://heltec.org/project/wifi-lora-32/

Both seems to work with the latest ESP32 micropython binary, but for the Heltec, the screen didn't answer. The command "i2c.scan()" didn't seems to work, it shows me all possible adresses (instead of just the OLED one).

I precise the Heltec one with his screen works perfectly with examples on Arduino IDE. I will make more tests soon.

Do you have any idea how to resolve my issue ? Do I have to open a new thread for this ?

peterhinch, 2019-10-01:
My first approach would be to check for i2c pullup resistors. However questions like this should be raised on [the forum](https://forum.micropython.org/) rather than in this bug tracker.

Koraze, 2019-10-05:
I did this before posting my issue with Pin.PULL_UP and PULL_DOWN commands, but none of these worked.

However, you were right. I used a 330 ohms pull_up resistor on SDA pin and it worked. I will ask the forum why the Pin.PULL_UP commande not worked in my case.

Thanks for your answer ;)

closed 2019-12-19 by dpgeorge

Harvzor, 2020-11-16:
In addition to what @Koraze said, checkout https://forum.micropython.org/viewtopic.php?t=4002.

The RST pin on the Heltec display needs to be set to 1.

```
rst = Pin(16, Pin.OUT)
rst.value(1)
```

PrimeTime416, 2022-04-15:
Setting the RESET pin to HIGH worked for me along with making sure your I2C pins match what's on the schematic for the OLED interface.

`from machine import Pin, SoftI2C
import SSD1306 as ssd1306
from time import sleep

i2c = SoftI2C(scl=Pin(15), sda=Pin(4))

oled_width = 128
oled_height = 64

rst = Pin(16, Pin.OUT)
rst.value(1)
oled = ssd1306.SSD1306_I2C(oled_width, oled_height, i2c)`

JBelmont72, 2024-02-23:
My headaches with this were solved when I switched my Adafruit OLED 128x64 to 5 Volts.  With 3.3 V I ws getting every version of error messages I have seen in this thread and a half dozen more.  For some reason, none of the many tutorials I read mentioned using 5 Volts.