micropython/micropython #4901: Declaring a specific `dict` causes the the attributes order of the namedtuple instance to change

ShenTengTu, 2019-07-07:
I use `dir()` to observe attributes of namedtuple instance.

**General** : 
```python
from collections import namedtuple
Test = namedtuple('Test', ['a', 'b', 'c'])
test = Test(a=2, b=4, c=6)
print(dir(test)) # output is ['__class__', 'a', 'b', 'c']
```
**Declare a specific `dict`** : 
```python
from collections import namedtuple
{'a':'c'} # occur problem
Test = namedtuple('Test', ['a', 'b', 'c'])
test = Test(a=2, b=4, c=6)
print(dir(test)) # output is ['__class__', 'a', 'c', 'b']
```

If key or value of `dict` is attribute name of namedtuple, the order would change.

Because Micropython not implement `namedtuple._fields`, so I use `dir()` to get it.

**Running firmware** : MicroPython v1.11 on 2019-05-29; ESP32 module with ESP32

dpgeorge, 2019-07-08:
There is not a problem here.  The return value of `dir()` does not need to be in order.  The named tuple's elements are still in the correct order which can be checked by doing `test[0], test[1], test[2]`.

ShenTengTu, 2019-07-08:
Thank reply, I fix the logical problem in my codes.

closed 2019-07-08 by ShenTengTu