micropython/micropython #4903: fs mount failed

dobotopensource, 2019-07-08:
The f_mount() could return FR_NO_FILESYSTEM somtimes which will reset the files of U disk!! I don't know why. The mcu is stm32f407, U disk space is about 100kB.
`// avoid inlining to avoid stack usage within main()
MP_NOINLINE STATIC bool init_flash_fs(uint reset_mode) {
    // init the vfs object
    vfs_fat = &fs_user_mount_flash;
    vfs_fat->flags = 0;
    pyb_flash_init_vfs(vfs_fat);

    // try to mount the flash
    FRESULT res = f_mount(&vfs_fat->fatfs);

    if (reset_mode == 3 || res == FR_NO_FILESYSTEM) {
        // no filesystem, or asked to reset it, so create a fresh one

        // LED on to indicate creation of LFS
        led_state(PYB_LED_GREEN, 1);
        uint32_t start_tick = HAL_GetTick();

        uint8_t working_buf[_MAX_SS];
        res = f_mkfs(&vfs_fat->fatfs, FM_FAT, 0, working_buf, sizeof(working_buf));
        if (res == FR_OK) {
            // success creating fresh LFS
        } else {
            printf("MPY: can't create flash filesystem\n");
            return false;
        }
`

dpgeorge, 2019-07-08:
> The f_mount() could return FR_NO_FILESYSTEM somtimes which will reset the files of U disk

Yes, this is correct behaviour: if there is no filesystem, or the filesystem became corrupt, then it will be automatically created.

dobotopensource, 2019-07-09:
What's the reason of file system corruption? Is there any way to avoid this ? The files of user were all cleared ...

dpgeorge, 2019-07-09:
> What's the reason of file system corruption?

There could be many reasons.  Eg power removed during a write, or contention with a PC doing writes to the USB MSC at the same time that the board writes to the filesystem.

closed 2025-02-04 by projectgus