micropython/micropython #4912: mDNS support for ESP32

goatchurchprime, 2019-07-09:
Discussion copied out from Roadmap thread here
https://github.com/micropython/micropython/issues/4821#issuecomment-509609186 

Things might have moved on since issue #2875  

What I said:

> I'd really like mDNS support in the socket.getaddrinfo() for looking up ".local" domains. (This is just the "easy" half of mDNS spec.)
> 
> Otherwise I've got to hard-code the ip-number when using mqtt to a local broker, which is not very robust as this number can change, which then requires taking down and reprogramming all my Micropython devices.
> 
> I've been getting around it using this minimal UDP sending code, which just about works on an ESP32. If this were done properly it would seriously help out with connecting embedded devices onto local networks in the real world where people don't always have access to the router to force it to use fixed ip-numbers.

What @dpgeorge said:

> Note that in the stm32 port mDNS queries and the responder are enabled for boards that have network interfaces.
> 
> For esp32 I think it's just a matter of enabling the feature.

My follow up comments:

> It would be perfect if gethostbyname() just worked for ".local" domains -- like it does in CPython -- so it's not a special case and people don't have to know what's going on. 
> 
> It's not obvious to the programmer whether this is a Micropython issue or a platform issue.  It would therefore really help out if we got an **"mDNS not implemented in this platform"** error for platforms where the feature is not enabled rather than just an empty list.  
> 
> Chasing down this rabbit hole ends up at [call to gethostbyname()](https://github.com/python/cpython/blob/e42b705188271da108de42b55d9344642170aa2b/Modules/getaddrinfo.c#L555) in CPython, and ends at a [call to dns_gethostbyname()](https://github.com/micropython/micropython/blob/3581deec816ff88a7dc2c0274b7c60b11e4ade34/extmod/modlwip.c#L1603) in Micropython.  
> 
> There's an [mDNS implementation](https://github.com/espressif/arduino-esp32/tree/master/libraries/ESPmDNS)  in arduino-esp32 land, and a lot of [mDNS stuff in the esp-idf](https://github.com/espressif/esp-idf/search?q=mdns&type=Code) (which I believe Micropython/ESP32 is built on).  

I don't know why it's not hooked up automatically, like it is in CPython.

dpgeorge, 2019-07-09:
> It would be perfect if gethostbyname() just worked for ".local" domains -- like it does in CPython

It's not really a CPython thing, it must be configured correctly on the PC in order to work.  Translating that to MicroPython, the TCP/IP stack must be configured to enable mDNS (or support added on to it in some way).

carstenblt, 2019-07-22:
There is an implementation available from loboris that could be ported/copied:

[https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/blob/fede6f610c9952c7242fc309c2bb4efc9afb0919/MicroPython_BUILD/components/micropython/esp32/network_mdns.c](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/blob/fede6f610c9952c7242fc309c2bb4efc9afb0919/MicroPython_BUILD/components/micropython/esp32/network_mdns.c)

dpgeorge, 2019-07-26:
See #4951 for an implementation.

dpgeorge, 2019-08-19:
mDNS on esp32 was added in 2ccf030fd1ee7ddf7c015217c7fbc9996c735fd1

closed 2019-08-19 by dpgeorge

george-hawkins, 2020-07-22:
Sorry to comment on a long-closed ticket. There's an interesting difference between the mDNS _responder_ implementation provided by @nickovs in his pure MicroPython [slimDNS](https://github.com/nickovs/slimDNS) library and the mDNS support added by @dpgeorge that calls down to the mDNS functionality provided by the ESP-IDF. The slimDNS implementation correctly includes the transaction ID of the query in the response whereas the corresponding ESP-IDF implementation does not. Some tools are forgiving of this but some common tools, like `dig`, are not. E.g. if query my board with `avahi-resolve` then everything works fine:

```
$ avahi-resolve --name espressif.local
espressif.local	192.168.0.248
```

Whereas, if I do the same with `dig` it treats the response as invalid due to its failure to echo back the query ID:

```
$ dig +short -p 5353 @224.0.0.251 espressif.local
;; Warning: ID mismatch: expected ID 39315, got 0
```

This is a shame, as working with names is generally nicer than having to use raw IP addresses. At the moment, I get different behavior depending on the system that I'm using - my Ubuntu system is forgiving when doing name resolution for Chrome but on my Android phone, the same address query is not resolved.

I've logged a bug about this against the ESP-IDF - see issue [#5574](https://github.com/espressif/esp-idf/issues/5574). So far it hasn't been taken up. If you come across this issue then please respond to the bug there and hopefully it'll get some more attention.