micropython/micropython #4915: Failed to build micropython for ports/unix ...

jiapei100, 2019-07-10:
Any suggestions?

```console
➜  unix git:(master) make V=1
python3 ../../py/makeversionhdr.py build/genhdr/mpversion.h
GEN build/frozen_mpy.c
python3 ../../tools/mpy-tool.py -f -q build/genhdr/qstrdefs.preprocessed.h build/frozen_mpy/upip.mpy build/frozen_mpy/upip_utarfile.mpy > build/frozen_mpy.c
Traceback (most recent call last):
  File "../../tools/mpy-tool.py", line 878, in <module>
    main()
  File "../../tools/mpy-tool.py", line 866, in main
    raw_codes = [read_mpy(file) for file in args.files]
  File "../../tools/mpy-tool.py", line 866, in <listcomp>
    raw_codes = [read_mpy(file) for file in args.files]
  File "../../tools/mpy-tool.py", line 720, in read_mpy
    raise Exception('incompatible .mpy version')
Exception: incompatible .mpy version
make: *** [../../py/mkrules.mk:119: build/frozen_mpy.c] Error 1
make: *** Deleting file 'build/frozen_mpy.c'
```

What's in **ports/unix/build/genhdr/mpversion.h**:

```code
// This file was generated by py/makeversionhdr.py
#define MICROPY_GIT_TAG "v1.10-539-ga73859d5a"
#define MICROPY_GIT_HASH "a73859d5a"
#define MICROPY_BUILD_DATE "2019-07-09"
```

Any suggestions?

Cheers

jiapei100, 2019-07-10:
Sorry, problem solved... Use the release version....

closed 2019-07-10 by jiapei100

dpgeorge, 2019-07-10:
You may need to rebuild mpy-cross (`make -C mpy-cross` from root dir), then `make clean` and `make` in the unix dir.