micropython/micropython #4916: trying to build mpy with dynamic libs for esp32 with xtensa-gcc

eminarcissus, 2019-07-10:
Hi,

I'm trying to adapt micropython dynamic lib for esp32 xtensa specific ports, everything went well except for when I trying to create the modx.elf file.

I'm getting a error like this

modx.o: In function `init':
modx.c:(.mpytext+0x5): dangerous relocation: l32r: literal placed after use: .mpytext.literal
modx.c:(.mpytext+0x24): dangerous relocation: l32r: literal placed after use: (.mpytext.literal+0x4)
modx.c:(.mpytext+0x38): dangerous relocation: l32r: literal placed after use: (.mpytext.literal+0x8)
modx.c:(.mpytext+0x55): dangerous relocation: l32r: literal placed after use: (.mpytext.literal+0xc)
Makefile:52: recipe for target 'modx.elf' failed
make: *** [modx.elf] Error 1

Any hints on what might be the cause of the problem? looks like the error is strongly related to mpextern.ld, if I do remove -T mpextern.ld it links but ofcourse objcopy will create nothing to output here. May I know what's behind the scene to create the mpy file?

dpgeorge, 2019-07-10:
> I'm trying to adapt micropython dynamic lib for esp32 xtensa specific ports,

I guess you are referring to #1627 ?  That PR is quite dated and pretty much superseded by #4535, which was merged.  Eventually (hopefully soon) there will be support to build .mpy files from C code, of which there is a simple example in #4535 (those bits were not merged).

eminarcissus, 2019-07-10:
thx 👍 , just noticed that from PR list, the old version finally builds, and it crashed when imported, I will try this one instead.

eminarcissus, 2019-07-10:
Tried with the latest code, after compile and import native_ex I got "ValueError: incompatible .mpy file", any ways to debug on failures like this?

robert-hh, 2019-07-10:
Did you also rebuild mpy-cross?

eminarcissus, 2019-07-10:
yes, I did finally get it built but failed to work.
Building problem seems like esp32 ports doesn't  have emit xtensa set to 1 as default.
Also it did came with a lot of problem except for that,
MICROPY_NLR_SETJMP setting seems like have some problem with emitglue.c's nlr_push/nlr_pop. 
Also when it compiles and starts, the native_ex module loads but failed to call, which crashes immediately.

closed 2019-07-12 by eminarcissus