micropython/micropython #4921: "FATAL: uncaught exception" on KeyboardInterrupt during large printout at normal REPL

mikewadsten, 2019-07-12:
This is a bug that I originally found in our own commercial product running a fork of MicroPython, but I have since reproduced on the mainline STM32 port as well.

At the friendly REPL, if you enter an expression such as `"a" * 10000` and then very quickly hit Ctrl-C, MicroPython will crash with "FATAL: uncaught exception <pointer>", where `<pointer>` corresponds to `mp_state_ctx.vm.mp_kbd_exception`.

I am able to reproduce this with both the v1.10 and v1.11 branches, including the current master (154062d9c). For v1.11 reproduction, I am using an STM32F415ZG based dev board internal to our company. My other reproduction is on our Silicon Labs EFM32 based product. The fact that I can reproduce on STM32 tells me the bug is not in our port.

I am _not_ able to reproduce on v1.9, at least where I've tested (including a microbit).

From my own analysis of the "uncaught exception" message, I believe the cause of the crash is that the REPL runs `parse_compile_execute` to execute the given expression, which sees the KeyboardInterrupt being raised, restores context, and re-raises -- but there's no surrounding `nlr_push` at the REPL, so `nlr_jump_fail` ends up being called.

dpgeorge, 2019-07-15:
Thanks for the report.

> there's no surrounding nlr_push at the REPL, so nlr_jump_fail ends up being called.

Yes that's pretty much it.  In detail it looks like the following is happening:
1. In `lib/utils/pyexec.c:parse_compile_execute`, the call to `mp_call_function_0()` executes the code and it takes a while to execute (print the large string), and ctrl-C comes in during this execution.
1. The VM does not check for the pending ctrl-C because the print of the string was the last thing it needed to do and it returns with an exception still pending.
1. Back in `parse_compile_execute`, `mp_hal_set_interrupt_char(-1)` is called to disable interrupts but there's still the `KeyboardInterrupt` pending.
1. When the REPL resumes and calls `mp_hal_stdin_rx_chr()` this function checks for pending exceptions and raises it.
1. There is no surrounding NLR handler so the exception is uncaught.

The problem exists for all ports that use `pyexec.c` (bug confirmed also on esp32).

Note that it's not enough to fix point (2) and make the VM always check for pending exceptions on return, because there's always the chance an exception comes in betwee the VM returning and execution of `mp_hal_set_interrupt_char(-1)`.

One option to fix it is to make `mp_hal_set_interrupt_char(-1)` also cancel/clear any pending exceptions.  That would mean such exceptions are lost.  The other option is to make the REPL loop check explicitly for pending exceptions and have code to deal with them (for a `KeyboardInterrupt` basically just do nothing).

dpgeorge, 2019-07-15:
See related #1667, #1722, #1738

dpgeorge, 2019-07-15:
> One option to fix it is to make `mp_hal_set_interrupt_char(-1)` also cancel/clear any pending exceptions.

A variation on this would be for this function to first set the interrupt char to `-1` (to stop any more incoming events), then if there is a pending exception raise it.

dpgeorge, 2020-02-07:
Fixed by 5a91cd9ff3e97b74ad6c9a615f810b8fcebbe110

closed 2020-02-07 by dpgeorge