micropython/micropython #4923: BUG: sys.modules['sys'] not pointing __import__('sys')

pmp-p, 2019-07-16:
Probably the same reason as :
https://github.com/micropython/micropython/issues/4427

dpgeorge, 2019-07-16:
This is because built-in modules are not registered in `sys.modules`

pmp-p, 2019-07-16:
ok then, but as they should, is it fixable at the point where import should could call __import__ instead of builtins__import  ?

dpgeorge, 2019-07-16:
> is it fixable at the point where import should could call `__import__`

Not really.  It can be implemented in the actual import implementation.  But if `__import__` could be overridden then the custom version of import could populate `sys.modules` itself.

pmp-p, 2019-07-16:
> But if __import__ could be overridden then the custom version of import could populate sys.modules itself.

i think override could not deal with ```__main__``` at startup, before hooking. eg:
```
import sys
sys.modules.get('__main__') # <= nope 
````
so it should be implemented in the actual import implementation