micropython/micropython #4924: stm32F769 discovery: linker FLASH_APP size

hoihu, 2019-07-16:
I stumbled across this in the stm32F769 discovery board definitions:

https://github.com/micropython/micropython/blob/master/ports/stm32/boards/STM32F769DISC/f769_qspi.ld#L20

I believe this is a copy&paste error? Shouldn't it read ```length = 1920k``` since the internal FS is activated (text0 starts at 0x08020000)?

dpgeorge, 2019-07-17:
Yes indeed it looks like a mistake.

chrismas9, 2019-07-17:
`FLASH_APP   .isr_vector`
`FLASH_APP (rx)  : ORIGIN = 0x08020000, LENGTH = 2016K   /* sectors 1-11 3x32K 1*128K 7*256K */`

If isr is in FLASH_APP why doesn't FLASH_APP start at 0x08000000?
Also 0x08020000 is sector 4, not sector 1 as in comment.

dpgeorge, 2019-07-17:
> If isr is in FLASH_APP why doesn't FLASH_APP start at 0x08000000?

Because this board (with this linker file) is configured to use mboot, so it's ISR table starts at 0x08020000.

> Also 0x08020000 is sector 4, not sector 1 as in comment.

Ok, that also needs fixing.

dpgeorge, 2019-07-17:
Fixed by 02b2ad4fbdbfae3e06ee4d0c686be36509d53d7d

closed 2019-07-17 by dpgeorge