micropython/micropython #4925: main.py not executing ESP32

smith-scot, 2019-07-16:
main.py is not being executed after power cycle.  I have only two files loaded boot.py and main.py.  My evidence that main is not being executed is as follows.  The last line in boot.py sets a variable to True.  The first line in main.py sets a different variable to True.  After power cycling the board and logging into the REPL, and running the dir() command, the variable set in boot.py is output to the screen but not the variable from main.py.  If this is not the correct place to raise this issue please let me know.  Thanks.

dpgeorge, 2019-07-17:
A better place to ask this would be at https://forum.micropython.org

Otherwise, it's best not to rely on `dir()` (although it should work).  Instead try doing a `print(...)` in main.py to check if it's being executed or not.

smith-scot, 2019-07-17:
Thanks Damien, I have ask this question in the forum.  Working through some of the suggestions now.  The problem with the print statements is that everything (boot.py and main.py) seems to work fine when I reset the board form the REPL (already logged in serially).  The problem comes when I cycle power without being logged in, I am depending on a LED for indication.  boot.py works in this situation but not main.py

cu, 2019-07-23:
@smith-scot Can you link to the thread? I'm having a similar issue.

jimmo, 2019-07-23:
> @smith-scot Can you link to the thread? I'm having a similar issue.

https://forum.micropython.org/viewtopic.php?f=18&t=6690

cu, 2019-07-23:
I had a similar issue getting main.py to actually run on boot. This may not be the same issue as OP describes but the symptoms were the same, so I'm recording it here for posterity.

The problem ended up being something timing-related with the sensor I'm reading from. I have a DHT22 temperature/humidity sensor and am trying to read from using MicroPython's built-in `dht` module. My `main.py` looked like this:
```python
import dht
import machine
import time

d = dht.DHT22(machine.Pin(4))

while True:
  d.measure()
  print('temp: %d, humidity: %d' % (d.temperature(), d.humidity()))
  time.sleep(1)
```
The symptoms were that when the board had power applied, some gibberish was printed and the loop stopped. No traceback or any other errors. The only way to get the loop to run was by issuing Ctrl+D in the REPL to do a soft reset, or by pressing the hard reset button on the board.

The fix is simple, I moved the `sleep()` to the top of the loop. I guess the DHT22 needs a few seconds to stablize, otherwise calling `measure()` too early results in a crash. The working code:

```python
import dht
import machine
import time

d = dht.DHT22(machine.Pin(4))

while True:
  time.sleep(1)
  d.measure()
  print('temp: %d, humidity: %d' % (d.temperature(), d.humidity()))
```

smith-scot, 2019-07-23:
I ended up not using main.py at all.  I received a suggestion in the forum to call a file (my_module.py in my case) from boot.py.  I then handled all the logical operations in my_module.py.  I still don't really understand why I can't get main.py working

peterhinch, 2019-07-24:
I'm puzzled by these reports. I can't replicate this on the reference board (ESP32-devkit-C) running MicroPython V1.11. main.py runs after a power cycle.

smith-scot, 2019-07-24:
I've curious all along if there was something going on with my board, or if
some how in the creation of main.py I made an error.  However I do not
having any problem calling by another name and I suspect that if I
explicated called main.py it would work (but I haven't tried that)

smith.randallscott@gmail.com
(205) 614-0096
www.LinkedIn.com/in/RandallScottSmith
<http://www.linkedin.com/in/RandallScottSmith>

peterhinch, 2019-07-25:
If you want to replicate my test, create **main.py** on the target filesystem with the following contents
```python
import time
while True:
    print('loop')
    time.sleep(1)
```
Check that it works by issuing a soft reset (ctrl-d).
Then disconnect the USB cable, wait a few seconds, and reconnect it.
If you fire up a terminal program you should see the output.

robert-hh, 2019-07-25:
Hello Peter, If you look at the related thread in the forum (link above), you see that a lot was tried. The error occurs when powered stand-alone. Since picocom plays with the modem control lines on startup, it may force a reset of a "espressif" type board. So I made a test main.py, which toggles the on-board led , and powered a few of my devices with my desk power supply and a cable definitely only having Vcc and GND connected. All of them worked as expected. The  code is in the thread.  I also added a machine.deepsleep(5000) at the end to force a reboot cycle.
I assume that is a incomplete reset issue or a hardware defect, which causes the device to behave like in REPL raw mode. That could be verified by powering up the device with and w/o pushing reset. But with a device that works fine that is hard to tell.

dpgeorge, 2019-08-19:
I'll close this as a hardware issue.  Feel free to reopen if software is suspected.

closed 2019-08-19 by dpgeorge

iotegrityio, 2021-11-25:
I’ve had exactly the same problem with ESP8266 running MicroPython v1.17. I used DHT11 sensor and my sleep was on the end of the code. I’ve added `sleep(1)` before initializing DHT11 in main and the problem was solved. I didn’t change anything else, just adding sleep after main.

It looks like when running cold boot or power cycle, the sensor cannot be initialized and we get an error. I was also able to an exception on my minicom terminal which helped me to pinpoint the issue. I’m pasting it here so that people coming can solve it. Think iit like a wisdom of ancients: https://xkcd.com/979/

Running code:

```
def main():
    log.info('starting main loop and sleeping initially')
—>    time.sleep(1)  # here we add sleep, be careful about the indentation.

    d = dht.DHT11(machine.Pin(config.DHT11_PIN))

    while True:
        led.blink(5, 50)
        d.measure()

        print("Sıcaklık: {}".format(d.temperature()))
        print("Nem: {}\n".format(d.humidity()))

        time.sleep(2)
```

Exception I’m getting after the power cycle (usb out, usb in). I can reproduce this behavior every time:

```
INFO - main - starting main loop and sleeping initially
Traceback (most recent call last):
  File "main.py", line 67, in <module>
  File "main.py", line 59, in main
  File "dht.py", line 17, in measure
OSError: [Errno 110] ETIMEDOUT

MicroPython v1.17 on 2021-09-02; ESP module with ESP8266
Type "help()" for more information.
>>> 
```

Looks like we can solve this in `dht.py` or we just add sleep after the main loop.

Do you still think this is a hardware issue, should we re-open it and investigate dht.py?

Regards,
Eren