micropython/micropython #4933: NoBackendError when libusb is installed via MacPorts

cdwilson, 2019-07-19:
I've got `libusb` installed via MacPorts and I'm getting a `NoBackendError` when using `pydfu.py` (`dfu-util` works fine):

```
$ make BOARD=STM32F4DISC deploy
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
GEN build-STM32F4DISC/genhdr/mpversion.h
CC ../../lib/utils/pyexec.c
CC moduos.c
LINK build-STM32F4DISC/firmware.elf
   text    data     bss     dec     hex filename
 321380      32   27632  349044   55374 build-STM32F4DISC/firmware.elf
GEN build-STM32F4DISC/firmware.dfu
Writing build-STM32F4DISC/firmware.dfu to the board
Traceback (most recent call last):
  File "../../tools/pydfu.py", line 591, in <module>
    main()
  File "../../tools/pydfu.py", line 570, in main
    init()
  File "../../tools/pydfu.py", line 99, in init
    devices = get_dfu_devices(idVendor=__VID, idProduct=__PID)
  File "../../tools/pydfu.py", line 424, in get_dfu_devices
    custom_match=FilterDFU(), **kwargs))
  File "/Users/chris/.anyenv/envs/pyenv/versions/default/lib/python3.6/site-packages/usb/core.py", line 1263, in find
    raise NoBackendError('No backend available')
usb.core.NoBackendError: No backend available
make: *** [deploy] Error 1
```

I think this is because pyusb uses `ctypes.util.find_library` to find the backend libs, which doesn't search `/opt/local/lib`.  However, it will search `DYLD_FALLBACK_LIBRARY_PATH`, so adding the following hack to `pydfu.py` fixes the problem:

```
if sys.platform == "darwin":
    import os
    os.environ["DYLD_FALLBACK_LIBRARY_PATH"] = "/opt/local/lib"
```

It seems like there should be a better way to fix this, since this seems like more of an issue with pyusb, rather than micropython.  However, I'm not quite sure how to tell pyusb (or `find_library()`) to look in `/opt/local/bin`.

Anybody else come up with a better way to resolve this?

andrewleech, 2019-08-06:
Is macports even still supported? You may get better results with homebrew.

cdwilson, 2019-08-07:
> Is macports even still supported?

Supported by MicroPython or supported in general?  AFAIK, MacPorts is still actively supported and I've been using it for years because it's been working for me without any issues (until this one).  Just wanted to see if there were any other fellow MacPorts users that had resolved this issue easily.

I'll take another look at Homebrew, might be time for me to switch! (I had originally stuck with MacPorts for some of the reasons outlined in http://terrychay.com/article/macports-vs-homebrew.shtml)

closed 2019-08-07 by cdwilson