micropython/micropython #4936: stm32 inline assembler: odd behaviour with memoryview argument

peterhinch, 2019-07-20:
I'm not sure if this is a bug or a limitation which should be documented.

If a memoryview into a bytearray is created and a slice passed to an assembler function, the function sometimes gets an incorrect address. The workround is explicitly to pass the address of the slice (using **uctypes.addressof**).

Unfortunately my attempts to create a trivial test case failed: in each case passing the slice worked. Consequently the assembler function is more involved than I'd have liked. The salient points are the reported address values and the fact that, in the failing case, the output array is only updated once (and then incorrectly).

```python
from uctypes import addressof

@micropython.asm_thumb
def lcopy(r0, r1, r2, r3):
    push({r0})  # Save address for test
    mov(r7, 0)  # Black
    mov(r5, 1)
    label(BYTE)  # Next byte
    mov(r6, 0x80)  # r6 == bit mask
    ldrb(r4, [r0, 0])  # Current source byte
    label(BIT)  # Next bit
    tst(r4, r6)  # Test masked bit
    ite(eq)
    strh(r7, [r1, 0])  # 0
    strh(r3, [r1, 0])  # 0xdead
    sub(r2, 1)
    beq(DONE)
    add(r1, 2)  # Next dest halfword
    lsr(r6, r5)  # mask >>= 1
    bne(BIT)
    add(r0, 1)
    b(BYTE)
    label(DONE)
    pop({r0})  # Restore address of arg 0 for test

def bar():
    dest = bytearray(20)
    source = b'\x0f\x80\xff\x00\x00\x00'
    mv = memoryview(source)
    for x in range(3):
        d = mv[2*x : 2 + 2*x]
        addr = lcopy(d, dest, 9, 0xdead)  # Fails
#        addr = lcopy(addressof(d), dest, 9, 0xdead)  # Works
        print(hex(addr), hex(addressof(d)), dest)
bar()
```
Correct output using addressof: addresses match, contents of dest are different on each pass.
```
0x200039d0 0x200039d0 bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\xad\xde\xad\xde\xad\xde\xad\xde\xad\xde\x00\x00')
0x200039d2 0x200039d2 bytearray(b'\xad\xde\xad\xde\xad\xde\xad\xde\xad\xde\xad\xde\xad\xde\xad\xde\x00\x00\x00\x00')
0x200039d4 0x200039d4 bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
```
Failing case: addresses don't match, contents of dest are incorrect and unchanging:
```
0x20003750 0x200039d0 bytearray(b'\xad\xde\x00\x00\x00\x00\xad\xde\x00\x00\xad\xde\x00\x00\x00\x00\xad\xde\x00\x00')
0x20003880 0x200039d2 bytearray(b'\xad\xde\x00\x00\x00\x00\xad\xde\x00\x00\xad\xde\x00\x00\x00\x00\xad\xde\x00\x00')
0x200039c0 0x200039d4 bytearray(b'\xad\xde\x00\x00\x00\x00\xad\xde\x00\x00\xad\xde\x00\x00\x00\x00\xad\xde\x00\x00')
```
The code treats the source array as 16 bit words and tests the most significant 9 bits of each word. If a bit is set it writes 0xdead to the dest array, otherwise 0 (as 16 bit words).

jimmo, 2019-07-20:
Both `addressof` and convert-to-register-for-non-basic-type (i.e. `convert_obj_for_inline_asm`) work via the buffer protocol. I believe the issue here is that `addressof` doesn't care if it's a readonly buffer or not, but `convert_obj_for_inline_asm` requires it to be writeable.

`memoryview(source)` is readonly, because source is `bytes`. So the fallback case is that it returns the pointer (to the `memoryview` object).

So `addressof` seems like a good workaround (or make `source` a `bytearray` instead), but possibly like you say a documentation update is in order. I couldn't find any documentation that described the conversion-to-register behaviour.

But... perhaps the requirement that it has to be a writeable buffer could to be revisited?

FWIW, I was able to repro with this case. It prints different values for `addr` and `addressof(d)`. Changing `source` to be a `bytearray`, or using `asm_fun(addressof(d))` makes them the same.

```
from uctypes import addressof

@micropython.asm_thumb
def asm_fun(r0):
    # No-op, just return r0 unchanged
    label(DONE)

def bar():
    source = b'\x00'
    mv = memoryview(source)
    d = mv[0:1]
    addr = asm_fun(d)
    print(hex(addr), hex(addressof(d)))

bar()
```

peterhinch, 2019-07-20:
Well spotted - I was unaware that a writeable buffer is assumed.

On the face of it this seems an unexpected requirement. If the requirement must be retained it would be good if a read-only buffer caused an exception to be raised. Currently it fails silently with a risk of the user code scribbling on invalid locations.

dpgeorge, 2019-07-25:
> On the face of it this seems an unexpected requirement.

I agree.  I'm happy to change it so that it works with read-only buffers (as well all read-write).

There's a similar constraint for viper type conversion in `mp_native_from_obj()` that would also need to be changed for consistency.

dpgeorge, 2019-08-05:
See #4976 for a fix.

dpgeorge, 2019-08-06:
Fixed by cd35dd9d9a29836906acdce60c931f6352b536d0

commit cd35dd9d (2019-08-06): py: Allow to pass in read-only buffers to viper and inline-asm funcs.

Fixes #4936.
closed 2019-08-06 by dpgeorge