micropython/micropython #4942: ESP32: RMT Neopixel implementation

carstenblt, 2019-07-22:
Loboris has implemented a Neopixel driver in C using the ESP32 RMT interface.

[https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/blob/master/MicroPython_BUILD/components/micropython/esp32/machine_neopixel.c](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/blob/master/MicroPython_BUILD/components/micropython/esp32/machine_neopixel.c)

I have been using it because it is quite a bit faster when controlling a larger number of LEDs. It also offers some convenient functions to work with the HSB color space.

I have copied it to upstream micropython without any problems.

Are you willing to include this implementation? If yes, I would open a pull request. The question would then be how you would like to have it integrated:

- just add it additionally
- replace the current implementation with a python wrapper to keep the API unchanged and have the new functionality available under a different namespace
- rewrite the C module to mimick the current API. Rename or remove the additional functionality.

dpgeorge, 2019-07-25:
The Loboris port is known to have had some GPL code in it, so it would be too risky to have any code from there come into this repository.

carstenblt, 2019-07-25:
I can't comment on that other that all the relevant files have the MIT licence header and are marked to be written by himself.

dpgeorge, 2019-07-25:
But it's hard to know if such code was/wasn't influenced by GPL code, if there's other GPL code around in the repository.  In this case it's better to be cautious and consider his code to be unusable here.

Since implementing an RMT driver is not that difficult it can be done for this repository as a fresh, clean implementation, and then there are no concerns about licensing.

carstenblt, 2019-07-27:
I didn't want to argue ;-)

closed 2019-07-27 by carstenblt

thijstriemstra, 2020-01-19:
> I have been using it because it is quite a bit faster when controlling a larger number of LEDs. It also offers some convenient functions to work with the HSB color space.

> I have copied it to upstream micropython without any problems.

@carstenblt i'm new to micropython ecosystem, how did you integrate it in upstream? a simple copy paste where? any other files i need to modify?