micropython/micropython #4949: vfs/fat code not working outside micropython

carstenblt, 2019-07-25:
I've been trying to implement threaded C code with file access and I believe that I am failing because it is relying on a micropython function in [vfs_fat_diskio.c](https://github.com/micropython/micropython/blob/master/extmod/vfs_fat_diskio.c):

`mp_call_method_n_kw(2, 0, vfs->readblocks);` in `disk_read()`

At least that is the call where I get the panic. I do not fully understand all of the code yet, I'd appreciate any help. I don't even know, what function it is calling yet but it should be replaceable.

dpgeorge, 2019-07-26:
The VFS code is not really intended to be used "outside MicroPython", mainly because the underlying block device could be written in Python.

But it still could be made to work, using the `FSUSER_NATIVE` option (grep for that symbol to see example usage).

carstenblt, 2019-07-26:
Thanks a lot! The answer was right above...

In case anyone is interested, this made it work, with stolen code from the stm32 ftp-server implementation:

`#include "extmod/vfs_fat.h"
#include "extmod/vfs.h"
#include "lib/oofatfs/ff.h"
#include "esp_spi_flash.h"

mp_uint_t storage_read_blocks(uint8_t *dest, uint32_t block_num, uint32_t num_blocks) {
    return spi_flash_read((block_num * 4096) + 0x200000, dest, num_blocks * 4096);
}

STATIC FATFS *lookup_path(const char **path) {
    mp_vfs_mount_t *fs = mp_vfs_lookup_path(*path, path);
    if (fs == MP_VFS_NONE || fs == MP_VFS_ROOT) {
        return NULL;
    }
    // here we assume that the mounted device is FATFS
    fs_user_mount_t *vfs = (fs_user_mount_t*)MP_OBJ_TO_PTR(fs->obj);
    vfs->flags |= FSUSER_NATIVE;
    vfs->readblocks[2] = MP_OBJ_FROM_PTR(storage_read_blocks);
    //FATFS *fatfs = &vfs->fatfs;
    //(char *)(fatfs->win) = win;
    return &vfs->fatfs;
}

char filepath[] = "testfile";
FIL fd;
FATFS *fs = lookup_path(&myfilepath);
FRESULT res = f_open(fs, &fd, myfilepath, FA_READ);`

closed 2019-07-26 by carstenblt