micropython/micropython #4952: Automatic stack overflow checking

dhalbert, 2019-07-27:
https://blog.japaric.io/stack-overflow-protection/ describes how Rust puts the stack at the bottom of RAM (with heap at the top) providing an automatic way to check stack overflow. It requires linking twice. It haven't looked at it in detail but it looks interesting.

dpgeorge, 2019-07-29:
That's not a bad idea.  And since we usually fix the size of the stack (eg 8k) it won't need to be linked twice, just move the stack to the start of RAM.

embeddedt, 2019-08-02:
I assume that this would only work well on single-threaded MicroPython ports?

dpgeorge, 2019-08-06:
> I assume that this would only work well on single-threaded MicroPython ports?

It would work on bare-metal ports that have at least one thread (the main thread).  It would only be there for additional safety, there would still need to be explicit stack checking for Python recursion overflow.