micropython/micropython #4962: ESP32 uploading issue with large files

dukeduck1984, 2019-07-30:
Hi,

I had a hard time uploading some large files (~60KB - 350KB) which are part of my Vue.js front end to the ESP32 board.  Have tried ampy, mpfshell and ftp server (with lobo's firmware), the uploading always ended up with some errors, the file was only partially uploaded and corrputed.

Is there any other upload tools that I can try? or is there a work around?

Thanks,
Kaiyuan

dpgeorge, 2019-07-30:
You can try using the `tools/pyboard.py` script in this repository, eg: `pyboard.py --device /dev/ttyUSB0 -f cp source_file :`

MrSurly, 2019-07-30:
Have you tried webrepl? Also, are you certain your FS isn't actually corrupt?

dukeduck1984, 2019-07-30:
@dpgeorge Thanks Damien, I will give it a try.  How can I upload a folder with all the files in it?

@MrSurly Thanks Eric.  No, I haven't tried webrepl as loboris fork doesn't come with the webrepl built-in.  I'm not sure if the FS is corrupted or not, but it seems more related to the file size.  For smaller files, e.g. the python files, although sometimes ampy does come up with an error,  eventually the files will be uploaded after multiple tries (I believe a lot of people reported similar issue with ampy).  However for bigger files (larger than 50kb) it's extremely difficult to get it uploaded.  What can I do to ensure the FS isn't corrupt?  Should I reflash the firmware, or is there a way to format the FS?

dpgeorge, 2019-07-30:
> How can I upload a folder with all the files in it?

You need to upload each file separately, also use `-f mkdir <dir>` to make directories.

> loboris fork

If the issues all occur when using that fork then you'll need to file an issue with at that repository.

robert-hh, 2019-07-30:
Using Filezilla, the v1.11-180 build of Micropython on ESP32 and my FTP server for ESP32, I have no problems uploading large files (~1.5 MByte) and complete directory structures. The only thing I had to change at FileZilla is the timeout, and that only for a very large file (78MByte -> SD Card, PYBD).

dukeduck1984, 2019-07-30:
@robert-hh It sounds very promising!  I will definitely try using your FTP server.  Would you pls share your FileZilla timeout setting?  The largest file I need to upload is under 400KB

robert-hh, 2019-07-30:
I changed the timeout setting from 20 to 100. The units seem to be seconds, and it was only needed for the 78MByte file.
P.S.: I gave up having a device with the Loboris fork, since it was not maintained any more. I anyhow only had it for comparison. The file uftpd.py may not work on the loboris firmware.

dukeduck1984, 2019-07-30:
@robert-hh Understood, I will give it a try anyway.  I'm also thinking about moving back to the main fork of MP ESP32 port as the same reason as yours.

dukeduck1984, 2019-07-30:
It seems that this issue might have something to do with the Loboris firmware - after multiple failure, I decided to flash the firmware of the micropython main fork here, and with @robert-hh 's uftpd, it worked out like a charm!  For sure uftpd will be my go-to uploading tool for esp32.  Thanks a lot @robert-hh !

Also thank you guys @dpgeorge @MrSurly for your time and efforts for the community!

I will close this issue.

Kaiyuan

closed 2019-07-30 by dukeduck1984