micropython/micropython #4963: framebuf module : Add support for rotation

iommu, 2019-07-31:
Add support for setting the rotation a frame buffer in the micropython framebuf module.

Currently the orientation of a display is limited by it's physical rotation. The best method I found to rotate a display is to use the [adafruit circuitpython framebuf](https://github.com/adafruit/Adafruit_CircuitPython_framebuf/blob/master/adafruit_framebuf.py) implementation instead of the native micropython one as it does allow for setting rotation.

peterhinch, 2019-07-31:
The Adafruit implementation performs operations in Python where the official version uses C so I'd expect it to be very much slower.

iommu, 2019-07-31:
This is true, using it I can see it's a lot slower but it's still a missing feature from the native lib. 

[From the forums](https://forum.micropython.org/viewtopic.php?t=6319#p36067) the best native solution people are currently recommending is to create 2 framebuffers (one rotated, one not) and then to write one on top of the other using the blit() function. Generally memory is a fairly scarce resource so this is far from an optimal solution.

MrSurly, 2019-07-31:
> is to create 2 framebuffers (one rotated, one not) and then to write one on top of the other

Why not 2 framebuffers pointing to the same block of RAM?

peterhinch, 2019-07-31:
I don't think this would work in the general case where a pixel occupies more than one byte (e.g. RGB565).

MrSurly, 2019-07-31:
> I don't think this would work in the general case where a pixel occupies more than one byte (e.g. RGB565).

Hmm, yeah, seems it needs a rotational transform for the actual coordinates ... but I haven't thought about this too deeply =P

dhalbert, 2019-07-31:
A bit of a diversion but in CircuitPython we have replaced `framebuf` with `displayio`, which takes a rather different approach, to support a wide variety of displays. Rotation is included, and lots of other stuff. https://circuitpython.readthedocs.io/en/latest/shared-bindings/displayio/__init__.html

peterhinch, 2019-08-01:
Arguably a full solution to rotation should include the three possible rotated directions (I have had a request for upside down display on one of my graphics libraries).

There are a range of useful enhancements to framebuf some of which which have been discussed e.g. https://github.com/micropython/micropython/issues/2973. I would like to see further graphics primitives such as circles and triangles with filled variants which could readily be implemented using standard algorithms. I would code these if I thought a PR might be approved. I suspect framebuf does not have a high priority; my offer to implement a "micro" solution to https://github.com/micropython/micropython/issues/2973 has gone unanswered.

andrewleech, 2019-08-06:
OpenMV has a wide range of image manipulation functions like this already available in C.

Their rotation function for instance does handle 3 dimensions.

You can use a (slightly out of date) version of their imaging library as a C user module with mainline python as described here: https://gitlab.com/alelec/mp_omv

Or perhaps it's a good time to work with them to bring some of their great image buffer work into mainline, as an optional module.