micropython/micropython #4964: Initial a WDT

Walkline80, 2019-07-31:
I am running some code in a new thread using While True. I feed the WDT every 2 seconds in this loop, but after the timeout period set by the WDT, the device will restart. I don't understand, I need a watch dog. But what I got is a timed restart device?

Walkline80, 2019-07-31:
The firmware I am using is: MicroPython v1.11-115-g592f68449 on 2019-07-04; ESP32 module with ESP32

dpgeorge, 2019-07-31:
Can you provide some minimal code which reproduces the problem?

jimmo, 2019-07-31:
The issue seems to be that it's being fed from a different thread. Can't reproduce if I create the WDT and feed it from the REPL but if you feed it from a thread it fails:

```python
import machine
import time
import _thread

w = machine.WDT(timeout=5000)

def thread_fn():
    while True:
        time.sleep_ms(1000)
        print('feed')
        w.feed()

_thread.start_new_thread(thread_fn, ())
```

I see

```
>>> feed
feed
feed
feed
...resets
```

jimmo, 2019-07-31:
Reading https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/wdts.html it sounds like this is the expected behaviour -- tasks (threads) have to feed their own watchdogs.

dpgeorge, 2019-07-31:
Thanks for the test case.

Our machine.WDT is not really specific about whether it is a global WDT or a per-thread WDT, because this topic never came up before.

So at least it needs some clarification around that point.  And should we try to make it a global WDT instead (if that's possible)?

Walkline80, 2019-07-31:
> The issue seems to be that it's being fed from a different thread. Can't reproduce if I create the WDT and feed it from the REPL but if you feed it from a thread it fails:
>
> ```python
> import machine [...]

@jimmo you're right, that's the thing what I have done, but I'm using class instead.....

But there's another question, according to source code of machine.WDT, WDT's id must be 0, so we have only one WDT in every piece of board, is this right?

[image]

Walkline80, 2019-07-31:
> Thanks for the test case.
> 
> Our machine.WDT is not really specific about whether it is a global WDT or a per-thread WDT, because this topic never came up before.
> 
> So at least it needs some clarification around that point. And should we try to make it a global WDT instead (if that's possible)?

@dpgeorge see my last reply, it seems like we have only one WDT to use, so I think....a global WDT is  necessary and usefully :)

dpgeorge, 2019-07-31:
>  it seems like we have only one WDT to use

It looks like there's one WDT for each thread.  Eg:
```python
import _thread, machine, time
def f():
    w = machine.WDT()
    while 1:
        print('feed')
        w.feed()
        time.sleep(1)
_thread.start_new_thread(f, ())
```
This won't crash.  But if the main thread starts its own WDT and doesn't feed it then it will.  So there are multiple WDT's (it's a software WDT not a hardware one).

closed 2019-08-01 by Walkline80