micropython/micropython #4966: I think I found a show-stopper bug.

jdcrunchman, 2019-08-01:
I defined a class User,   in the "www" directory as follows.

In file user.py (see attached ZIP)

class User:
    def __init__(self, name):
        self.userName = name
        self.otherUsersInRange = []
        self.signal_level = 0
        self.public_key = ''
        self.uniqueID = 'zasdertghyuirdcfd145498bpl'

    def getname(self):
        return (self.userName)

    def setSignalLevel(self, signal):
        self.signal_level = signal  ## Level is in DB

    def getSignalLevel(self):
        return (self.signal_level)

    def getRemoteUsers(self):
        ## return a list of other users within range
        return(self.otherUsersInRange)

    def numUsersInRange(self):
        return len(self.otherUsersInRange)

    def getPubKey(self):
        return (self.public_key)

    def getUniqueID(self):
        return (self.uniqueID)

## I've also defined this class.   in a file "webinput.py",   also in the www directory
and included in the attached ZIP.

from www.user import User    <---- even though I'm running this module from www
## it seems to need to implicitly tell where the user file is located.

## I'm using PyMakr with Atom IDE.

class Message:
    def __init__(self, owner):
        self.msg_text = ''          ## text of the message
        self.msg_owner = owner      ## owner of the message

    def getMessageText(self):
        return(self.msg_text)

class WebInput:
    def __init__(self):
        self.users = []
        self.myMessage = 0

    def AddUsers(self, user_list):
        ## Gets one or more users from the remote ESP32
        self.users = user_list

    def return_user_n(self, usernumber):
        if usernumber == 0:
            return('')
        else:
            name = self.users.getname()
            return(name)

    def numUsers(self):
        return (self.users.count())

### . and now I do the following...

wInput = WebInput()    <--- I instantiate a WebInput object.

user1 = User('adam')    <--- Just instantiated 3 users
user2 = User('Jason')
user3 = User('Andy')

usr_list = []       <---- create a blank user list

usr_list.append(user1)   <---- append the 3 users into "usr_list" 
usr_list.append(user2)
usr_list.append(user3)

wInput.AddUsers(usr_list)   <---- add this list to my wInput

num = usr_list.count()          <---- Here I want to just see how many are in the usr_list
                                                <---- this is line 52 here where I get the error.

>>> Running webinput.py

>>>
>>> Traceback (most recent call last):
  File "<stdin>", line 52, in <module>
TypeError: function takes 2 positional arguments but 1 were given

I tried to drag and drop the .py files,  but this does NOT support the file types.
[wfiles.zip](https://github.com/micropython/micropython/files/3458536/wfiles.zip)

These are the two files.    From a virgin Micropython system,  put these in the www directory
and run or import the webinput.py this requires no other dependencies.

I'm using this MicroPython image:   
esp32-20190727-v1.11-180-g8f55a8fab.bin

The latest.

There is ABSOLUTELY NO WAY this should't work.

Please check it out.

Email me at jdcrunchman@gmail.com

dhalbert, 2019-08-01:
This code is calling `count()` incorrectly.
```
num = usr_list.count() 
```
You want `len(usr_list)` to find out the number of elements in `usr_list`. `usr_list.count(x)` does something different: it counts how many times `x` appears in `usr_list`. You cannot omit the argument.

https://forum.micropython.org is a good place to ask support questions.

jdcrunchman, 2019-08-02:
I never did find anything about how to use it,    it seemed obvious to me
that it would seem to me to be that "count" was
how many items in the array.

John

MrSurly, 2019-08-02:
> I never did find anything about how to use it

It's a standard method of [Python lists](https://docs.python.org/3/tutorial/datastructures.html).

If you can't find MicroPython specific documentation WRT a language feature, then it's probably following the Python 3 spec.

closed 2019-08-05 by dpgeorge