micropython/micropython #4968: memory malloc error on stm32f407

WingCoo, 2019-08-02:
print error log:
```
#Malloc: 0x20013560 16
#Malloc: 0x20013570 12
#Malloc: 0x200135A0 16
#Malloc: 0x200135B0 24
**_#Malloc bytes:4096 failed_**
#Testing 1kB...
Addr:0x20013B10
#Testing 1.25kB...
Addr:0x20013B10
#Testing 1.5kB...
Addr:0x20013B10
#Testing 2kB...
**_Addr:0x2001B5B0_**
#Testing 3kB...
###Malloc bytes:3072 failed
```
source code: 
```
void *m_malloc(size_t num_bytes) {
    void *ptr = malloc(num_bytes);
    if (ptr == NULL && num_bytes != 0) {
        //PrintInfo("num_bytes : %d ", num_bytes);
        SEGGER_RTT_printf(0, "#Malloc bytes:%d failed\r\n", num_bytes);
        SEGGER_RTT_printf(0, "#Testing 1kB...\r\n");
        ptr = malloc(1024);
        if (!ptr) {
            SEGGER_RTT_printf(0, "###Malloc bytes:%d failed\r\n", 1024);
        } else {
            SEGGER_RTT_printf(0, "Addr:0x%08x\r\n", ptr);
            free(ptr);
        }
        SEGGER_RTT_printf(0, "#Testing 1.25kB...\r\n");
        ptr = malloc(1250);
        if (!ptr) {
            SEGGER_RTT_printf(0, "###Malloc bytes:%d failed\r\n", 1250);
        } else {
            free(ptr);
            SEGGER_RTT_printf(0, "Addr:0x%08x\r\n", ptr);
        }

        SEGGER_RTT_printf(0, "#Testing 1.5kB...\r\n");
        ptr = malloc(1536);
        if (!ptr) {
            SEGGER_RTT_printf(0, "###Malloc bytes:%d failed\r\n", 1536);
        } else {
            free(ptr);
            SEGGER_RTT_printf(0, "Addr:0x%08x\r\n", ptr);
        }
        SEGGER_RTT_printf(0, "#Testing 2kB...\r\n");
        ptr = malloc(1900);
        if (!ptr) {
            SEGGER_RTT_printf(0, "###Malloc bytes:%d failed\r\n", 1900);
        } else {
            free(ptr);
            SEGGER_RTT_printf(0, "Addr:0x%08x\r\n", ptr);
        }
        SEGGER_RTT_printf(0, "#Testing 3kB...\r\n");
        ptr = malloc(3072);
        if (!ptr) {
            SEGGER_RTT_printf(0, "###Malloc bytes:%d failed\r\n", 3072);
        } else {
            free(ptr);
            SEGGER_RTT_printf(0, "Addr:0x%08x\r\n", ptr);
        }

        m_malloc_fail(num_bytes);
    }
#if MICROPY_MEM_STATS
    MP_STATE_MEM(total_bytes_allocated) += num_bytes;
    MP_STATE_MEM(current_bytes_allocated) += num_bytes;
    UPDATE_PEAK();
#endif
    // PrintInfo("malloc %d : %p", num_bytes, ptr);
    SEGGER_RTT_printf(0, "#Malloc: 0x%08x %d\r\n", ptr, num_bytes);
    return ptr;
}
```
heap & stack address:
```
.heap           0x20012f94     0x4000 load address 0x080b2a44
                0x20012f94                . = ALIGN (0x4)
                0x20016f94                . = (. + _minimum_heap_size)
 *fill*         0x20012f94     0x4000 
                0x20016f94                . = ALIGN (0x4)

.stack          0x20016f94      0x800 load address 0x080b2a44
                0x20016f94                . = ALIGN (0x4)
                0x20017794                . = (. + _minimum_stack_size)
 *fill*         0x20016f94      0x800 
                0x20017794                . = ALIGN (0x4)
```

dpgeorge, 2019-08-06:
I'm not sure what the issue is that's being reported here.

The best way to report an issue is to use an unmodified version of the MicroPython code and provide a test script that shows the problem.

zhjc2014, 2019-08-06:
> I'm not sure what the issue is that's being reported here.
> 
> The best way to report an issue is to use an unmodified version of the MicroPython code and provide a test script that shows the problem.

The ram of stm32f407 is 192kB, and the ram of stm32f429 is 256kB. I have set the heap end at addr 0x2001f000/0x2002f000 on both chip, the max size of the xxx.py file which can execute correctly is about 4kB/15kB. More file size will cause the memory allocation error, is it normal, or some error has happened ?

dpgeorge, 2019-08-07:
> the max size of the xxx.py file which can execute correctly is about 4kB/15kB

It depends what the total heap size is.  The heap runs from `_heap_start` to `_heap_end`, and `micropython.mem_info()` will report the size of the heap and other memory statistics.

If the heap is only 16k then the size of scripts that can be compiled and executed will be limited.

zhjc2014, 2019-08-07:
I think the heap of stm32f429 is large enough, it's 80kB, but the max size of the script is just about 15kB...
```
.heap           0x2001279c    0x14000 load address 0x080b2d4c
                0x2001279c                . = ALIGN (0x4)
                0x2002679c                . = (. + _minimum_heap_size)
 *fill*         0x2001279c    0x14000 
                0x2002679c                . = ALIGN (0x4)

.stack          0x2002679c      0x800 load address 0x080b2d4c
                0x2002679c                . = ALIGN (0x4)
                0x20026f9c                . = (. + _minimum_stack_size)
 *fill*         0x2002679c      0x800 
                0x20026f9c                . = ALIGN (0x4)
```

dpgeorge, 2019-08-07:
You need to check the output of `micropython.mem_info()` to get the actual size of the heap.

To make further progress on this please provide full steps to reproduce the error on one of the officially supported boards (see `ports/stm32/boards` directory).

zhjc2014, 2019-08-07:
This is the mem info printed blow.
```
stack: 612 out of 3072
GC: total: 114048, used: 65008, free: 49040
 No. of 1-blocks: 922, 2-blocks: 152, max blk sz: 822, max free sz: 111
stack: 612 out of 3072
GC: total: 114048, used: 65040, free: 49008
 No. of 1-blocks: 924, 2-blocks: 152, max blk sz: 822, max free sz: 111
stack: 612 out of 3072
GC: total: 114048, used: 65040, free: 49008
 No. of 1-blocks: 924, 2-blocks: 152, max blk sz: 822, max free sz: 111

```
This is my script blow, if I add several lines more(e.x. 10 lines), the script won't execute success. 
```
#devType: ML
#isUsingRail: True
import micropython
import rtt
import gc
micropython.alloc_emergency_exception_buf(100)
dType.SetQueuedCmdClear(api)
dType.SetQueuedCmdStopExec(api)
dType.SetQueuedCmdStartExec(api)
dType.SetCPParams(api,100.0,100.0,100.0, 1)
dType.SetPTPJointParams(api,200.0,200.0,200.0,200.0,200.0,200.0,200.0,200.0, 1)
dType.SetPTPCoordinateParams(api,200.0,200.0,100.0,100.0, 1)
dType.SetPTPJumpParams(api,20.0,100.0, 1)
dType.SetPTPCommonParams(api,50.0,50.0, 1)
dType.SetPTPLParams(api,200.0,200.0, 1)
dType.SetLostStepParams(api,5.0, 1)
for i in range(0, 99):   
   gc.collect()
   micropython.mem_info()
   dType.SetPTPCmdEx(api,1 ,142.6265, -114.3997, 62.5038, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,181.6254, -134.7993, 62.5041, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,142.6265, -114.3997, 62.5038, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,181.6254, -134.7993, 62.5041, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,142.6265, -114.3997, 62.5038, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,181.6254, -134.7993, 62.5041, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,142.6265, -114.3997, 62.5038, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,181.6254, -134.7993, 62.5041, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,142.6265, -114.3997, 62.5038, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,181.6254, -134.7993, 62.5041, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,142.6265, -114.3997, 62.5038, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,181.6254, -134.7993, 62.5041, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,142.6265, -114.3997, 62.5038, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,181.6254, -134.7993, 62.5041, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,142.6265, -114.3997, 62.5038, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,181.6254, -134.7993, 62.5041, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,142.6265, -114.3997, 62.5038, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,181.6254, -134.7993, 62.5041, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,142.6265, -114.3997, 62.5038, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,181.6254, -134.7993, 62.5041, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,142.6265, -114.3997, 62.5038, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,181.6254, -134.7993, 62.5041, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,142.6265, -114.3997, 62.5038, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,181.6254, -134.7993, 62.5041, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,181.6254, -134.7993, 62.5041, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,142.6265, -114.3997, 62.5038, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,181.6254, -134.7993, 62.5041, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,142.6265, -114.3997, 62.5038, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,181.6254, -134.7993, 62.5041, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,142.6265, -114.3997, 62.5038, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,181.6254, -134.7993, 62.5041, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,142.6265, -114.3997, 62.5038, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,181.6254, -134.7993, 62.5041, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,142.6265, -114.3997, 62.5038, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,181.6254, -134.7993, 62.5041, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,142.6265, -114.3997, 62.5038, -38.7328, 1)
[... 114 lines cut ...]
   dType.SetPTPCmdEx(api,1 ,181.6254, -134.7993, 62.5041, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,142.6265, -114.3997, 62.5038, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,181.6254, -134.7993, 62.5041, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,181.6254, -134.7993, 62.5041, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,142.6265, -114.3997, 62.5038, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,181.6254, -134.7993, 62.5041, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,142.6265, -114.3997, 62.5038, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,181.6254, -134.7993, 62.5041, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,142.6265, -114.3997, 62.5038, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,181.6254, -134.7993, 62.5041, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,142.6265, -114.3997, 62.5038, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,181.6254, -134.7993, 62.5041, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,142.6265, -114.3997, 62.5038, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,181.6254, -134.7993, 62.5041, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,142.6265, -114.3997, 62.5038, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,181.6254, -134.7993, 62.5041, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,181.6254, -134.7993, 62.5041, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,142.6265, -114.3997, 62.5038, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,181.6254, -134.7993, 62.5041, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,142.6265, -114.3997, 62.5038, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,181.6254, -134.7993, 62.5041, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,181.6254, -134.7993, 62.5041, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,142.6265, -114.3997, 62.5038, -38.7328, 1)
   dType.SetPTPCmdEx(api,1 ,181.6254, -134.7993, 62.5041, -38.7328, 1)
```

zhjc2014, 2019-08-07:
The allocation info below, is printed by the function "gc_dump_alloc_table()".
```
 0> 
 0> 00000: h=hhhhFFLhTSBFh==h====FFLhTFTFTFTh=Fh==MDBTh====h=======FFLhTFFT
 0> 00400: FTFTFTFh=FShSh=Shh======h=Shh=======hh=======h=====h============
 0> 00800: ====Sh=h==========h=====================hLhTFTFTh=FFLhTFFTFTFTFT
 0> 00c00: FTFFLhFFLhTFTFTFTFTFTFTFTFTFFLhFFLhTFTFTFTFTFTFTFTFTFFLhFFLhTFTF
 0> 01000: TFTFTFTFTFTFTFFLhFFLhTFTFTFTFTFTFh================h=============
 0> 01400: ==============h=================================================
 0> 01800: ================================================================
 0> 01c00: ================================================================
 0> 02000: ================================================================
 0> 02400: ==============h====================h============================
 0> 02800: ========================================h=======================
 0> 02c00: =================h============================================TF
 0> 03000: TFTFFLhFFLhTFTFTFTFTFTFTFTFTFFLhTh=======h=hFh=hTh=hFh===hTh====
 0> 03400: hFh====hTh====hFh=h=h=STh===hFh=====hTh===hFh===hTh===hFh======h
 0> 03800: =Th====hFh===hh=SFh===========hLh===hhh======h=Th===hFh=========
 0> 03c00: =hTh=======h=====hFh===hTh=====hFh===hTh===hFh=====hTh===hFh====
 0> 04000: ==h=Th===hFh======h=Th===hFh======h=Th===hFh======h=Fh===hLh====
 0> 04400: ==h==hh=======h====hFh======h==Fh=======h====hLh===========h=hh=
 0> 04800: =====hTh======h=Fhh=======h===Th================h=Fh======h==Th=
 0> 04c00: ===hFh======h=Th===hFh===========h=Th======hFh======h=Th===hFh==
 0> 05000: ====h=Thh=======h===Fh======h=Th================h==Fh=====hTh===
 0> 05400: hFh======h=Fh===hLh======h=hh======h=Fh===hFh======h=Lh======h=h
 0> 05800: h===hTh======h=Fh=======h======h=Th======h=Fh===hTh=====hFh=====
 0> 05c00: =h=Th======h=Fh===hTh======h=Fh===hTh======h=Fh===hTh===hFh=====
 0> 06000: =h=Th===hFh=====hThh=======h===Fh=====hFh===hLh=====hhh===hFh===
 0> 06400: hFh===hLh===hhh===hTh===hFh===hTh===hFh===h=Th=======h===hFh===h
 0> 06800: =Th===hFh===h=Th===hFh===h=Th===hFh===========h=Th========hFh===
 0> 06c00: ===h=Th======hFh=====hTh=====hFh=hFh=hLh=hhh=h=Fh==h==Fh=h=Lh=h=
 0> 07000: hh=h=Th=h=FFLhTFh=h=Fh=h=Th=h=Fh=h=Th=h=Fh=h=Th=Fh=======h=h=h=T
 0> 07400: h==h==Fh=hTh=h=Fh=hTh=h=Fh=h=Th=h=FTFDDDDDFLShh===hTFh==h===hh==
 0> 07800: =Dh=====h===Dh=h==h===h=h===Dh=Dh===Dh=h===Dh===h===hh===Dh=DDDh
 0> 07c00: ====Dh====h===Dhh===Dh=h===h===h===h===h===hh===Dhh===BBBBBBBBBB
 0> 08000: BBh===BBBBBTB=BFB=BTB=BBBBBFB=BTB=BFB=BTB=BFB=BTB=BFB=BTB=BFB=BT
 0> 08400: B=FB=BFB=BLB=BhB=BFB=BFB=LB=BBhB=BTB=FB=BTB=FB=BTB=FB=TB=BFB=TB=
 0> 08800: FB=BTB=FB=BBTB=BBBBBBBBBBBBBBBBBBBBBBFB=BTB=BFB=TB=BFB=FBB=LB=hB
 0> 08c00: =FB=FB=LB=hB=TFTFB=TB=FB=TB=FB=TB=FB=TB=FB=BTB=FB=TB=FB=Th======
 0> 09000: ================================================================
 0> 09400: =============B=h====FFLhhFFLh==hTFTSFh==========================
 0> 09800: ================================================================
 0> 09c00: ================================================================
 0> 0a000: ================================================================
 0> 0a400: ================================================================
 0> 0a800: ================================================================
 0> 0ac00: ================================================================
 0> 0b000: ================================================================
 0> 0b400: ================================================================
 0> 0b800: ================================================================
 0> 0bc00: ================================================================
 0> 0c000: ================================================================
 0> 0c400: ================================================================
 0> 0c800: ===========================FFFFLhTFLhTFTDFBTTFh====h==h======TFT
 0> 0cc00: FTFTFTFFLhFFLhTFTFTFTFFFFFFFFFTFTFTFTFTFFLhFFLhTFTFFTFTFTFTFTFFF
 0> 0d000: TFTFFLhFFFFFFFFFFFLhTFTFTFFFTFTFTFTFTFFFTFFLhTFFFFFFLhTFTFFFTFTF
 0> 0d400: TFTFFFTFTFFLhFFLhTFTFTFTFTFFTFTFTFTFFFFFLhTFTFTFFFFFLhTFFTFFTFTF
 0> 0d800: TFTFFFFFLhFFLhTFTFFFFFTFTFTFTFFTFTFTFFLhFFFFFLhTFTFFFFFFLhFFLhTF
 0> 0dc00: TFFFFTFTFTFTFTFTFTFFLhFFLhTFTFTFTFTFFTFTFTFTFFFFFLhFFLhTFTFFFFFT
 0> 0e000: FTFTFTFFTFTFTFFLhFFFFFLhTFTFTFFFFFTFTFTFTFFTFTFFLhTFFFFFFLhTFTFT
 0> 0e400: FFFFFTFTFTFTFFTFFLhTFTFFFFFFLhTFTFTFFFFFTFTFTFTFFFLhFFLhTFFFFTFT
 0> 0e800: FTFTFTFFFFFTFTFTFFLFhFFLhTFTFFFFTFTFTFTFTFFFFFTFTFFLhTFFFLhTFFTF
 0> 0ec00: FFFTFTFTFTFTFFFFFTFFLhTFTFFFFFFFFFLhTFFTFFTFTFTFTFTFFFFFFLhTFTFT
 0> 0f000: FFFLhTFFTFFFFTFTFTFTFFFFFFLhFFLhTFTFFTFTFTFTFFFFTFTFTFFLhFFFFFFL
 0> 0f400: hTFTFTFFTFTFTFTFFFFTFTFFLhTFFFFFFFLhTFTFTFFTFTFTFTFFFFTFFLhTFTFF
 0> 0f800: FFFFFLhTFTFTFFTFTFTFTFFFFFLhFFLhTFFFFFTFTFTFTFTFFTFTFTFFLFFFhFFL
 0> 0fc00: hTFTFFFFFTFTFTFTFTFFTFTFFLhTFFFFFLhTFFTFFFFFTFTFTFTFTFFTFFLhTFTF
 0> 10000: FFLhTFFTFTFTFTFTFFTFFLhTFTFFFFFLhTFTFFFFFFTFTFTFTFFFFTFFLhTFTFFF
 0> 10400: LhTFFTFFFFTFTFTFTFTFFFFFFLhTFTFTFFFLhTFFTFFFFTFTFTFTFFFFFFLhFFLh
 0> 10800: TFTFFFLhTFTFTFFFFFLhTFFTFFFFFTFTFTFTFFFLhFFLhTFTFFFFTFTFTFTFFFFF
[... 6 lines cut ...]
 0> 12400: LhTFTFTFFFTFTFTFTFFFLhFFLhTFFFFTFTFTFTFTFFFFFTFTFTFFLFhFFLhTFTFF
 0> 12800: FFTFTFTFTFTFFFFFTFTFFLhTFFFLhTFFFLFFFhFFLhTFTFFFFFTFTFTFTFTFFTFT
 0> 12c00: FFLhTFFFFFLhTFFTFFFFFTFTFTFTFTFFTFFLhTFTFFFFFLhTFFTFFFFFTFTFTFTF
[... cut ...]
 0> 1a400: ================================================================
 0> 1a800: ================================================================
 0> 1ac00: ===================h============================================
 0> 1b000: ================================================================
 0> 1b400: ================================================================
 0> 1b800: =======.........................................................
 0> 1bc00: ............................
 0> 
```