micropython/micropython #4970: sys.version_info should be compatible with CPython

mattytrentini, 2019-08-03:
`sys.version_info` differs in structure from [CPython's definition](https://docs.python.org/3/library/sys.html#sys.version_info). This can be a small stumbling block when porting ([example](https://github.com/mattytrentini/QR-Code-generator/tree/master/python)).

In MicroPython, `version_info` is a tuple of three components, in CPython it's a class with subscripting to access the _five_ components. As well as subscripting, the components can be accessed by name (eg `sys.version_info.major`). 

Presumably the MicroPython structure for `version_info` was chosen for lower memory use but it seems that the small cost in memory may be worthwhile for compatibility.

sponster-au, 2019-08-04:
I'm going to work on this issue at the sprint; at least we'll know what the memory cost of the change are.

sponster-au, 2019-08-05:
Further discussion with Damien: Many platforms don't want to spend the bytes on this feature. Having it available on some platforms is worse than the having it available on no platforms.

The current situation - making sys.version_info available as a tuple means that it is still straightforward to determine the version if that is required.

mattytrentini, 2019-08-06:
I'm not sure how many extra bytes it costs (?) - but that's reasonable.

How about we leave this ticket open but change the title to indicate that the difference from CPython should be documented? 

Specifically, [version_info documentation](https://docs.micropython.org/en/latest/library/sys.html#sys.version_info) should be updated to include something like "Note that this implementation differs from CPython. Only the first three version numbers are supported and they can be referenced only by index, not by name (`.major`, `.minor` etc)."

dpgeorge, 2019-08-06:
> How about we leave this ticket open but change the title to indicate that the difference from CPython should be documented?

A good place for such documentation would be `tests/cpydiff/`.

dpgeorge, 2019-08-15:
Documentation added in d5a77416064f8155b0a0817211541c296b49d2c7

closed 2019-08-15 by dpgeorge