micropython/micropython #4971: RFC: "REPL interact" as easier to implement debugger?

andrewleech, 2019-08-03:
Would it be possible/easy to create a `micropython.interact()` function that would break to repl in the current scope, with a `continue` function that can be run in said repl to resume the program flow after you've investigated what's going on?

This could be used like a hard coded breakpoint, say if your code detects an error it will throw you into the repl at that point so you can inspect what's going on.

MrSurly, 2019-08-03:
Something akin to `import pdb; pdb.set_trace()`?

andrewleech, 2019-08-04:
Yeah basically, without the significantly more complex trace infrastructure.

The idea was inspired by the new `debugger ()` function in 3.7

dpgeorge, 2019-08-06:
> Would it be possible/easy to create a `micropython.interact()` function that would break to repl in the current scope,

Yes, probably it wouldn't be too difficult, but it seems like a hack :)

> so you can inspect what's going on.

Note that MicroPython doesn't implement `locals()` so the usefulness of this "inspection REPL" would be limited.

> The idea was inspired by the new `debugger ()` function in 3.7

Do you mean `breakpoint()`? https://www.python.org/dev/peps/pep-0553/

> Something akin to `import pdb; pdb.set_trace()`?

I think it'd be better to go straight for this rather than add a MicroPython specific extension.  See #4265

andrewleech, 2019-08-06:
Ah yes, I forgot about the lack of locals, that would limit it. And yes, breakpoint() is what I meant.

I didn't realise there was a wip for trace of any kind, I agree that something more in line with official python is always going to be preferable!

dpgeorge, 2019-08-06:
> I didn't realise there was a wip for trace of any kind

You might want to look through it, it's very good, very faithful to the CPython interface.