micropython/micropython #4972: Python file can't import another py file

fei2020, 2019-08-05:
I embed javascript port into my emscripten application. mp_js_do_str works fine in my c code.
And that pyexec_file invoke a python file is fine too. i.e. upy works in web application. But The python file can't "import" another module which is also a python file in the same directory.
I test it in emscripten MEMFS root directory "/".

This is the gist below:
```
	FILE* fp = NULL;
	fp = fopen("t2.py", "w");
	fprintf(fp, "import woxiao\nwoxiao.HUDPrint(\"print by t2.py\")\n");
	fclose(fp);

	fp = fopen("t1.py", "w");
	//fprintf(fp, "import woxiao\nwoxiao.HUDPrint(\"print by t1.py\")\ntry:\nimport t2\nwoxiao.HUDPrint(\"+ import t2 done\")\npass\nexcept ImportError:\nwoxiao.HUDPrint(\"- import t2 failed\")\n");
	fprintf(fp, "import woxiao\nwoxiao.HUDPrint(\"print by t1.py\")\nimport t2\n");
	fclose(fp);

	//pyexec_file ("t2.py");  //works fine
	pyexec_file ("t1.py");  // can't import t2
```
ps. "woxiao.HUDPrint" is just a debug print function in my own "woxiao" module which is extended in my C code. And it works fine.

dpgeorge, 2019-08-05:
Importing files in the Javascript port is currently not imported.  It's needed to implement the `mp_lexer_new_from_file()` and `mp_import_stat()` functions in `ports/javascript/main.c`.

fei2020, 2019-08-09:
> Importing files in the Javascript port is currently not implemented. It's needed to implement the `mp_lexer_new_from_file()` and `mp_import_stat()` functions in `ports/javascript/main.c`.

Thank you. I'll try to follow your instructions.

fei2020, 2019-08-13:
> Importing files in the Javascript port is currently not implemented. It's needed to implement the `mp_lexer_new_from_file()` and `mp_import_stat()` functions in `ports/javascript/main.c`.

import py file works. But I not sure if my method is correct. Here's what I did:

1 mp_lexer_new_from_file() in lexer.c was enabled by set "#define MICROPY_READER_POSIX    (1)" in ports/javascript/mpconfigport.h. And comment the mp_lexer_new_from_file() in the ports/javascript/main.c.

2 Copy the implementation of mp_import_stat from ports/unix/main.c.

ps. I use upy 1.11.

2

pmp-p, 2020-05-09:
@fei2020 you can overload __import__ and write your own importer since https://github.com/micropython/micropython/pull/4929

The https://github.com/littlevgl/lv_micropython/tree/lvgl_javascript branche can show you how to do it with synchronous XHR