micropython/micropython #4973: esp32/machine_rtc: Make it easy to change the RTC UserMem size.

nevercast, 2019-08-05:
Hello,

I'm using a lot of RTC memory for my own C level code inside some C Modules. I'm running out of slow mem space when linking. I believe that this is my primary issue: https://github.com/micropython/micropython/blob/master/ports/esp32/machine_rtc.c#L54

While having user memory available from Python is a really nice feature, usually, it conflicts with my design.

I would like to be able to change MEM_USER_MAXLEN from the `make` line, as I prefer to not manipulate MicroPython source in my builds so that I can keep up to date with upstream improvements.

I believe if `#define MEM_USER_MAXLEN 2048` was replaced with
```
#ifndef MEM_USER_MAXLEN 
  #define MEM_USER_MAXLEN 2048
#endif 
```

Then I would be able to specify this value to be 0x0 or something sufficiently small on my `make` line when building the ESP32 port that it wont occupy a half my slow memory.

I'm happy to make a PR, but I thought this should be discussed first.

Josh.

nevercast, 2019-08-05:
I've made a PR proposing the exact change I mentioned in this issue. For ease of merge. I'm still open to discussing this, however.

MrSurly, 2019-08-05:
I think the idea is reasonable.

dpgeorge, 2019-12-19:
Looks like this was implemented by #5349

closed 2019-12-19 by dpgeorge