micropython/micropython #4986: MicroPython on the ESP32-D2WD (2MiB)

zrecore, 2019-08-07:
In case anyone else is interested, I've managed to mod the source and a partitions.csv file to get MicroPython and it's filesystem working on an ESP32-D2WD, which only as 2MiB of internal Flash, not 4MiB.

Modded `ports/esp32/modules/flashbdev.py`, `ports/esp32/modesp.c`, and `ports/esp32/partitions.csv`
[image]
[image]
[image]

I would recommend we not hard-code things to 0x200000 anymore, as Espressif offers 2MiB internal SPI flash chips. Perhaps we rework these changes a bit to make the built process account for flash user start addresses lower than 0x200000?

nevercast, 2019-08-07:
Nice one! 

I agree that this constants hurt. I'm starting to think it's becoming more and more preferable to have board configs for ESP32. We can still use sdkconfig if we make the ESP32 port an ESP IDF component and use the KConfig system. 

Then once we have a good place for them we can start moving this constants out to various configs.

zrecore, 2019-08-07:
See pull request #4987

tcpipchip, 2019-08-09:
I followed the process, but my NINA W102 (2MB flash) is resetting! 
Can you inform me where address to burn those 3 bin files created ? (application, partition and bootloader) if i use flash tools ?

zrecore, 2019-08-09:
> I followed the process, but my NINA W102 (2MB flash) is resetting!
> Can you inform me where address to burn those 3 bin files created ? (application, partition and bootloader) if i use flash tools ?

@tcpipchip when `make` is run, firmware.bin will be generated. Please see the following instructions.

With the changes shown in the PR, you will need to set the env variables.

Go into the `/ports/esp32` folder.
(Replace `/dev/ttyUSB0` with your actual port.)

if using a makefile which imports Makefile... adjust/add the following
```
PORT = /dev/ttyUSB0
FLASH_MODE = dio
FLASH_FREQ = 40m
## if using a 2MiB ESP32 board
FLASH_SIZE = 2MB
FLASH_USER_START = 0x120000 # Defaults to 0x200000 address if not provided
FLASH_STORAGE_PARTITION_SIZE = 0xA0000 # 640MiB max. Defaults to 0x200000 if not provided (2MiB partition on 4MiB Flash ESP32 boards)
PART_SRC = partitions-2MiB.csv # Use the 

```

or .envrc if using direnv
```
# .envrc file

export ESPIDF="$HOME/esp-idf"
export FLASH_SIZE="2MB"
export FLASH_MODE="dio"
export FLASH_FREQ="40m"
export FLASH_USER_START=0x120000
export FLASH_STORAGE_PARTITION_SIZE=0xA0000
export PART_SRC="partitions-2MiB.csv"
```
Run `make clean` then `make`

Before flashing, erase the flash, then flash the firmware.
```
esptool.py -p /dev/ttyUSB0 erase_flash
esptool.py -p /dev/ttyUSB0 write_flash -z 0x1000 build/firmware.bin
```

tcpipchip, 2019-08-09:
Thanks to the feedback @zrecore 

And those 3 files
ports/esp32/modules/flashbdev.py, ports/esp32/modesp.c, and ports/esp32/partitions.csv

Still Need to modify too ?

zrecore, 2019-08-09:
> Thanks to the feedback @zrecore
> 
> And those 3 files
> ports/esp32/modules/flashbdev.py, ports/esp32/modesp.c, and ports/esp32/partitions.csv
> 
> Still Need to modify too ?

@tcpipchip  You can see all file changes in the Pull Request, see https://github.com/micropython/micropython/pull/4987/files

tcpipchip, 2019-08-10:
Good Morning!!!!

Now NINA W102 is running Micropython!!!! Thanks @zrecore 

You are the best!!!!!!!!!!!!!!!

Btw we ported the AT COMMANDS to 2Mb ESP32 too! 

Did you do that too ?

zrecore, 2019-08-10:
Wow, that's excellent news!

I did not add any AT commands...that may be existing code.

Please note, I am in the process of working with the MicroPython community
to integrate feedback, so these code changes may yet be modified further.

tcpipchip, 2019-08-11:
Zrecore

Using 2mb version.Did you have problems with station mode to access the AP ?

I am getting this message after boot

OSError: STA required

tcpipchip, 2019-08-11:
Sorry, the message is
STA must be active
[image]

tcpipchip, 2019-08-11:
[image]

tcpipchip, 2019-08-17:
Hi, the behavior of your esp32 is going well using 2mb ? Mine is very strange, sometimes not boot, or not get wifi, or not open tcp sockets

zrecore, 2019-08-17:
This is an untested patch.

tcpipchip, 2019-08-17:
Yes! A lot of problems! I aborted to use it!

zrecore, 2019-08-24:
@tcpipchip for every issue you find, would you mind documenting:

 - What happened
 - What you expected
 - How to reproduce the issue

dpgeorge, 2019-09-10:
The D2WD is now supported by the GENERIC_D2WD board, via commit 80d37d936c9832dc4b69ad631ad07697e26764d9

closed 2019-09-10 by dpgeorge

tcpipchip, 2019-09-10:
i will try soon to see if solve the MQTT access with 3 certifications error

tcpipchip, 2019-09-10:
can you help em to solve that certification error ?

zrecore, 2019-09-10:
@tcpipchip I'll do my best to help you, but please create a new GitHub issue for the problem.