micropython/micropython #4988: DS18X20 rom code, how to convert bytearray to string and back?

dukeduck1984, 2019-08-08:
Hi,

How can I convert the ds18x20 rom code which is bytearray to string in order to list the devices nicely on the GUI?  Also how can I convert it back to bytearray?

Pls see below for what I have tried, but always got UnicodeError.   I could find much in the docs.

```
>>> ds1
bytearray(b'(\xaa\xec\x01\x19\x13\x028')
>>> ds1.decode()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
UnicodeError:
>>> ds1.decode('utf-8')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
UnicodeError:
>>> str(ds1, "utf-8")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
UnicodeError:
```

Thanks,
Kaiyuan

dukeduck1984, 2019-08-08:
Forgot to mention, it's the esp32 port.

MicroPython v1.11-182-g7c15e50eb on 2019-07-30; ESP32 module with ESP32

peterhinch, 2019-08-08:
This query would be better raised in [the forum](http://forum.micropython.org/index.php).

dukeduck1984, 2019-08-08:
Altough cumbersome, this is my solution so far...

···
class RomCodeConvert:
    @staticmethod
    def _to_romcode_string(bytearray_romcode):
        """
        e.g.：bytearray_romcode = bytearray(b'(\xaa4\xe4\x18\x13\x02;')
        after convertion：'28 aa 34 e4 18 13 02 3b'
        :param bytearray_romcode: bytearray
        :return: str
        """
        import ubinascii
        string_hex_list = [str(ubinascii.hexlify(bytes([el])), 'utf8') for el in bytearray_romcode]
        romcode_str = ' '.join(string_hex_list)
        return romcode_str

    @staticmethod
    def _to_romcode_bytearray(string_romcode):
        """
        e.g.：string_romcode = '28 aa 34 e4 18 13 02 3b'
        after convertion：bytearray(b'(\xaa4\xe4\x18\x13\x02;')
        :param string_romcode: str
        :return: bytearray
        """
        import ubinascii
        string_hex_list = string_romcode.split()
        bytes_list = [int.from_bytes(bytes(ubinascii.unhexlify(el)), 'big') for el in string_hex_list]
        romcode_bytearray = bytearray(bytes_list)
        return romcode_bytearray
···

closed 2019-08-08 by dukeduck1984