micropython/micropython #4990: ESP8266: 'micropython-schedule': [Errno 22] EINVAL, packages may be partially installed

dentex, 2019-08-09:
Hello,
On a nodeMCU ESP8266, I was trying to install this lib:
https://github.com/rguillon/schedule
but I get the error:

```>>> import upip
>>> upip.install("micropython-schedule")
Installing to: /lib/
Warning: micropython.org SSL certificate is not validated
Installing micropython-schedule 0.1.8 from https://files.pythonhosted.org/packages/5b/5a/2399b34a971405ffffd3db8c4190fe1f97c0c86d8f25eea066564c675647/micropython-schedule-0.1.8.tar.gz
Error installing 'micropython-schedule': [Errno 22] EINVAL, packages may be partially installed
```
version: `MicroPython v1.11-8-g48dcbbe60 on 2019-05-29`

Is there anything I can try?
Thanks.

dpgeorge, 2019-08-16:
It looks like the esp8266 doesn't have enough RAM to download and extract the tgz file.

Try downloading it to your PC then freezing it into the esp8266 firmware.

Otherwise, this issue is more suited to the forum: https://forum.micropython.org

closed 2019-08-16 by dpgeorge

dentex, 2019-08-16:
Thanks, this is exactly how I got it working.