micropython/micropython #4992: ESP32: how to stop sta_if reconnect

dukeduck1984, 2019-08-10:
Hi,

It seems that if I entered a wrong ssid or password, the sta_if will try to reconnect forever.  Is this a feature?  How can I stop that loop, or at least limit the number of retries?

Firmware: MicroPython v1.11-194-g3d02ebb4e on 2019-08-09; ESP32 module with ESP32

Thanks,
Kaiyuan

MrSurly, 2019-08-11:
What I usually do is tell it to connect, then enter a loop that checks if it's connected, but with a time limit.

dukeduck1984, 2019-08-11:
No, what I meant was the method ```sta.connect()``` itself would perform a reconnect loop which I didn't ask it to.  For example, below is what happened when I entered a wrong AP, it retried to connect about every 2 seconds, until I manually entered ```sta.disconnect()```

```
>>> I (44252) wifi: STA_DISCONNECTED, reason:201
no AP found
I (46692) wifi: STA_DISCONNECTED, reason:201
no AP found
I (49142) wifi: STA_DISCONNECTED, reason:201
no AP found
I (51592) wifi: STA_DISCONNECTED, reason:201
no AP found
I (54032) wifi: STA_DISCONNECTED, reason:201
no AP found
I (56482) wifi: STA_DISCONNECTED, reason:201
no AP found
I (58932) wifi: STA_DISCONNECTED, reason:201
no AP found
I (61372) wifi: STA_DISCONNECTED, reason:201
no AP found
I (63822) wifi: STA_DISCONNECTED, reason:201
no AP found
I (66262) wifi: STA_DISCONNECTED, reason:201
no AP found
wifi.staI (68712) wifi: STA_DISCONNECTED, reason:201
no AP found
.disconneI (71162) wifi: STA_DISCONNECTED, reason:201
no AP found
ct()
>>>
```

peterhinch, 2019-08-12:
The following script works as expected on the reference ESP32 board.

The specified LAN does not exist. It prints "connecting" for the timeout period, then prints "Timeout" and quits. If I substitute my actual WiFi credentials it prints "connecting" for about 5s before printing the network config details.
```python
import network
import time
sta_if = network.WLAN(network.STA_IF)
ap = network.WLAN(network.AP_IF) # create access-point interface
ap.active(False)         # deactivate the interface
start = time.ticks_ms()
to = 10000
while not sta_if.isconnected():
    if time.ticks_diff(time.ticks_ms(), start) > to:
        print('Timeout')
        break
    print('connecting to network...')
    sta_if.active(True)
    sta_if.connect('roger_rabbit', 'no_such_lan')
    while not sta_if.isconnected() and time.ticks_diff(time.ticks_ms(), start) < to:
        print('connecting')
        time.sleep_ms(500)
else:
    print('network config:', sta_if.ifconfig())
    a = sta_if.config('mac')
    print('MAC {:02x}:{:02x}:{:02x}:{:02x}:{:02x}'.format(a[0],a[1],a[2],a[3],a[4]))
```

dukeduck1984, 2019-08-12:
@peterhinch Thank you.  I figured out that the "loop" was actually debug info, so I disabled it with ```esp.osdebug(None)```

closed 2019-08-12 by dukeduck1984