micropython/micropython #4993: port-javascript: check list

pmp-p, 2019-08-10:
### easy to fix

* heap/stack init is not clear
mp_js_init(stack_size) with stack= parameter instead of heapsize=
 https://github.com/micropython/micropython/blob/master/ports/javascript/wrapper.js#L59
but in fact is heap https://github.com/micropython/micropython/blob/master/ports/javascript/main.c#L88

* browser example is minimalist and does not provide a default micropython web server and possibly a websockify example and a tool to launch test suite.

* builtins time.sleep and input should at least throw a warning/error instead of crashing browsers.

* an asyncio flavour should be provided as a frozen module.

* please use cpython C-API names when interfacing JS and C that will save everybody's time. either for understanding api or writing documentation see https://docs.python.org/3/c-api/veryhigh.html. that would ease transition later to https://github.com/pyhandle/hpy

### annoying things :

* passing script data for interactive or script should be separate functions, actually all use mp_js_do_str(text) and alloc context of text is not clear if pausing/resuming is needed. But those are not the same :
https://docs.python.org/3/c-api/veryhigh.html#c.PyRun_SimpleString
https://docs.python.org/3/c-api/veryhigh.html#c.PyRun_InteractiveLoop

* parser/lexer should work on a pre allocated buffer either for interactive or script mode, filled with stringToUTF8( text, pointer, max_size ) as a real C string.
  - why pre alloc ? because max size of code that interpreter can compile is fixed by heap limit anyway so save useless/unclear malloc/free operations beetween js/C
  - execution mode could be simply set at end of buffer in a # comment when using asynchronous execution.

* mp_js_do_str(text)  is not wasm compliant and won't work when using clang 10 from latest-upstream emscripten branch. so better implement the two previous point and just get rid of it.

### heavy lifting :

* micropython core should be switched to non nlr with https://github.com/micropython/micropython/pull/4131. emscripten-upstream gets seriously confused with setjmp.

* files fd, sockets fd, pipes and not only stdin/stdout should be multiplexed via stdin/stdout via a common interface for node / browser and worker. zero copy rpc would be better ( arrow / capnproto ? )

embeddedt, 2019-08-12:
>  emscripten-upstream gets seriously confused with setjmp.

This interests me. Could you provide more details?

pmp-p, 2019-08-12:
@embeddedt somewhere in those blocks https://github.com/emscripten-core/emscripten/projects/1#column-1392613
actually only no nlr can be compiled ( but it can display pystone results before crashing ! )
i can try to get the exact error GOT.func entry if you need it

pmp-p, 2019-09-23:
Updated the list of what could be sampled as :
 "what you get for reward of supporting a new Python platform"

To people giving enough interest to read that, thank you and note that :

This is not cpython : that list can fixed with good will and wasm is the future.

embeddedt, 2020-12-28:
#6738 is a significant problem for the JS port, as we are finding the GC to be quite unreliable, making it useful only for small examples.

jonnor, 2024-07-03:
I belive this has now been superseeded by the Webassembly port?

pmp-p, 2024-07-04:
> I belive this has now been superseeded by the Webassembly port?

Hi, asm.js (javascript) and webassembly are different targets, eg asm.js works on old browsers and node but wasm only work on cpu emulators and recent browsers. Both ports may lack a browser testsuite. The most promising port may be https://github.com/micropython/micropython/pull/13676 which will behave the same on **all** wasi runtimes (browser or not).

closed 2024-07-04 by pmp-p