micropython/micropython #4997: UART on ESP32 - panic error

beyonlo, 2019-08-10:
Hi all.

I'm using the Espressif DevKitC with ESP32-WROVER-B module.

I'm trying this code:

```
from machine import UART
uart = UART(1, 115200)

``` 

First line, the import, works fine. But the second line of code show this error and after micropython is rebooted: 

> I (31142) uart: ALREADY NULL
> Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
> 

**The complete output is this:**

>>> from machine import UART
>>> uart = UART(1, 115200)
I (335976) uart: ALREADY NULL
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400d3e80  PS      : 0x00060730  A0      : 0x800dd94e  A1      : 0x3ffc1af0  
A2      : 0x0000044d  A3      : 0x3ff49054  A4      : 0x00000009  A5      : 0xffffffff  
A6      : 0xffffffff  A7      : 0x00000012  A8      : 0x800d3e80  A9      : 0x3ffc1ad0  
A10     : 0xcccccccc  A11     : 0x00060723  A12     : 0x00060720  A13     : 0x00000000  
A14     : 0xffffffff  A15     : 0x00000012  SAR     : 0x00000017  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xcccccccd  LBEG    : 0x400972ac  LEND    : 0x400972b7  LCOUNT  : 0xffffffff  

Backtrace: 0x400d3e80:0x3ffc1af0 0x400dd94b:0x3ffc1b10 0x400df8d1:0x3ffc1b40 0x400dc381:0x3ffc1b60 0x400e95fd:0x3ffc1b80 0x400e05a4:0x3ffc1c20 0x400dc519:0x3ffc1c80 0x400dc546:0x3ffc1ca0 0x400f9a42:0x3ffc1cc0 0x400f9c9c:0x3ffc1d60 0x400f1714:0x3ffc1da0 0x40092e35:0x3ffc1dd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4928
ho 0 tail 12 room 4
load:0x40078000,len:10188
load:0x40080400,len:6736
entry 0x400806e8
I (501) spiram: Found 64MBit SPI RAM device
I (501) spiram: SPI RAM mode: flash 40m sram 40m
I (501) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (507) cpu_start: Pro cpu up.
I (511) cpu_start: Application information:
I (516) cpu_start: Compile time:     07:47:57
I (521) cpu_start: Compile date:     May 29 2019
I (526) cpu_start: ESP-IDF:          v3.3-beta1-268-g5c88c5996
I (533) cpu_start: Starting app cpu, entry point is 0x40082e48
I (524) cpu_start: App cpu up.
I (1398) spiram: SPI SRAM memory test OK
I (1399) heap_init: Initializing. RAM available for dynamic allocation:
I (1399) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1405) heap_init: At 3FFB9DD0 len 00026230 (152 KiB): DRAM
I (1412) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1418) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1424) heap_init: At 40097B1C len 000084E4 (33 KiB): IRAM
I (1431) cpu_start: Pro cpu start user code
I (107) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
MicroPython v1.11 on 2019-05-29; ESP32 module with ESP32
Type "help()" for more information.
>>> 

What is wrong?

Thank you so much.

robert-hh, 2019-08-10:
It fails if I used the code like you do, it works if I specify rx and tx, like in:
```
from machine import UART
uart=UART(1, 115200, tx=4, rx=5)
```
But anyhow, it should not crash.

beyonlo, 2019-08-12:
Hello @robert-hh 

Thank you for the reply. Now a tested with GPIO 25 and 26, and works =D. I have just the message **I (200692) uart: ALREADY NULL** - that is a normal message, right?

```
>>> from machine import UART
>>> uart=UART(1, 115200, tx=25, rx=26)
I (200692) uart: ALREADY NULL
>>> 
```

> Looking into the code, the default GPIO for UART1 is 9 and 10, and for UART2 is 16 and 17. GPIO 9 and 10 are used for the SPI flash, GPIO 16 and 17 for the SPI RAM. So the defaults must be changed.

Are you sure about GPIO 9 and 10? Becouse the ESP32 chip don't have the GPIO number 9 and 10. See picture attached - esp32_and_Espressif DevKitC-PINOUT.

[image]

Thank you so much.

robert-hh, 2019-08-14:
You have to look at the schematics of the Wrover module itself. The flash memory is inside that module.
The message "I (200692) uart: ALREADY NULL" is normal. I do not see it because I disabled debug messages.

dpgeorge, 2019-08-15:
Looks like there are a couple of issues here:
- logging of "ALREADY NULL" is confusing
- the default UART pins are no good for some modules

I think both should be fixed.  The latter would need the concept of esp32 boards so that the default UART pins can be configured at build time for the specific board.

kmanyu, 2019-09-01:
how can use UART in ESP32?

> Looks like there are a couple of issues here:
>
> * logging of "ALREADY NULL" is confusing
> * the default UART pins are no good for some modules
>
> I think both should be fixed. The latter would need the [...]

closed 2026-05-25 by dpgeorge